oliexdev/openScale #3: Publish on f-droid

Teyro, 2017-02-25:
Is there a chance to get this app on fdroid?

oliexdev, 2017-02-25:
Hi Teyro,

thanks for your suggestion. Sure is that possible, I opened a ticket to submit the openScale app see https://gitlab.com/fdroid/rfp/issues/51
But we have to wait until the review process is done ;)

Best Regards from Germany
Oli

Teyro, 2017-02-26:
Heyho,

Sry next text is in german ;)

Danke für deine schnelle Eintragung ;) Sach mal planst du die App weiter zu entwickeln / und oder zu verbessern? Diese ganzen Funktionen für Körperfett etc kann die MiScale ja garnicht :( Kann man das irgendwie ausblenden?

oliexdev, 2017-02-26:
Hi Teyro,

I will answer in English and German maybe other visitors are interested in your question as well. 
Yes I still developing the App so any ideas, suggestions or any help would be appreciated. :)
You can already hide unwanted data fields. Just disable for example body fat or water percentage under the settings. I hope this answers your question, if not don't hesitate to ask again.

Ich werde in Englisch und in Deutsch antworten, vielleicht interessiert sich ein Besucher auch für deine Frage.
Ja ich entwickle die App weiter, daher sind Ideen, Vorschläge oder Unterstützung herzlichst willkommen. :)
Du kannst bereits nicht gewollte Einträge ausblenden. Deaktiviere einfach in den Einstellungen zum Beispiel den Körperfettanteil oder den Wassergehalt. Ich hoffe das beantwortet dir deine Frage, wenn nicht sag einfach bescheid.

Best regards
Oli

oliexdev, 2017-02-28:
F-Droid added openScale so I close this thread

closed 2017-02-28 by oliexdev