oliexdev/openScale #4: feature request. more measurements

nutpantz, 2017-03-01:
Just found this on fdroid. Looks great.
Only thing I can see it I might need is more measurements.
Not that these measurements need to be tracked, just compared to the last or first measurements.
Measurements like
Biceps, forearms, chest, neck, thigh and calf maybe even left and right sides.
Great for weight loss to see what is going missing
And good for body sculpting to see where muscle is being built.

Great work.

oliexdev, 2017-03-02:
Hi nutzpantz

thanks for your suggestion. I am not quiet sure how is the best way to implement more measurements, so that the app looks not overloaded and confusing. Do you have any design ideas?
The next question is how many measurements are needed and which? Where to start and where to end? For example should the neck circumference included or not. Some people will need it, but most of them not. An idea is to add measurement fields dynamically where the user can freely give them a title, a unit and a valid range. But this is not easy to implement and again the app should be easy to use. Maybe you or somebody else have another idea?
I am sorry at the moment the best way is to use the comment field to note any further measurements. 

Best regards
Oli

nutpantz, 2017-03-03:
Hello
Personally I like a button that switches to a page just about measurements.
One side would show the body and approximately where to measure for each
It should show the last measurement taken next to the new measurement input spot
You could have the body in the middle and measurements on each side for left and right.
Maybe one or two on the bottom for custom entries.
Maybe a show/hide list in settings for people who just want to enter some measurements and not clutter the screen unnecessarily or remind them that they are not measuring everything.

As for what measurements are needed that's a personal thing and more for the developer as to who
He is aiming his app at.

Weight watchers benefit from the primary ones
(Thigh, waist, hips, chest, neck, biceps) as they change a lot and can encourage people when they can't see just the waist changing.

Body builders would need more 
(Calf, forearm, maybe wrist and ankles(I'm not sure myself))
The more options the more out will appeal to a broader range of users

Larry0ua, 2017-04-09:
Maybe you just need some custom user-provided field names, it would fit nicely into Measurements settings section

oliexdev, 2017-04-11:
@Larry0ua 
as I already mention above this would be a solution but also a complex one. Everything graph, table, export/import function, and the database itself and so on need to be dynamically programmable.
I make a small step in this direction because now the measurement views can be dynamically added... I guess this will take more time. 

Best regards
oli

martinler, 2018-01-14:
Hello, any idea about integration with Google Fit? Beurer stock app does not provide this feature, so i'm looking for some replacement. Thanks!

oliexdev, 2018-01-14:
@martinler 
If you had read the [FAQ](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#why-are-the-measurements-are-not-synchronise-to-a-cloud-eg-google-fit--garmin-connect--dropbox-and-so-on), you would have known that this not supported. 
Btw. to corrupt this issue with your different question is not very nice.

erijo, 2018-03-25:
I was going to post this as a reply to @oliexdev in #233, but I guess the conversation fits better here.

@oliexdev wrote:
> I have in my mind to have a separate tab page (in the data entry activity) for body measurements (e.g. thigh, waist, hip, chest, neck, biceps and so on) and scale measurements (e.g. weight, fat, muscle, water and so on). 

I'm thinking that the data entry activity isn't a big problem. The measurements are few enough for one to be able to scroll them easily. Introducing separate tabs might make it slightly more cumbersome to use.

I think the two big issues are the table and graph view. They need some work to handle the amount of measurements we have already today. For the table view I have been thinking about making table be horizontal instead of vertical. I started implementing this on a branch some weeks ago, but got side tracked. When I get some time I will try to publish some early draft of it to get some hands on experience and feedback.

For the graph view I haven't really thought much, but one idea would be to have multiple graphs under each other where similar measurements can be grouped together. Then you could easily scroll between them to get an overview of the current status. Another idea would be to plot fat, water and bone in the same graph as weight and have the lines be
1. water
2. water + bone
3. water + bone + fat
4. weight

That would give a nice overview of the current body composition.

oliexdev, 2018-04-08:
> I'm thinking that the data entry activity isn't a big problem. The measurements are few enough for one to be able to scroll them easily. Introducing separate tabs might make it slightly more cumbersome to use.

I am not sure if it's more cumbersome to have a tab menu. It would be more clearly and not overfilled and we have a better disjunction between those two groups. You could for example for the body measurements show a stylized body picture, disable/enable one group and so on or could handle measurements slightly differently.

For the graph I don't see a big problem you could easily configure your graph view and it is easy to change it "on the fly". I want to stick with it.
(But the statistics overview could be improve a lot but that is another story)

For the table view to implement two sub tabs would reduce also the overall columns but could still lead for to much columns in one tab for small devices. I am not really understand your idea to use horizontal table instead of vertical?

oliexdev, 2018-05-14:
Close with #264

closed 2018-05-14 by oliexdev