oliexdev/openScale #6: Inappropriate scale for charts

taraska7, 2017-03-03:
First of all, thanks for the app!

I'd like to mention that the scale you use while building charts (starting from 0) isn't the best solution. This results in almost straight line, especially on small screen of a smartphone. This makes the chart completely useless. No one will weight 0 or 10 or 20 (even 60 for me personally) kilos. And there's no need to use this values while building charts. The better way is to take upper and lower results, add some margin (say, 2-5 kg) and use the resulting scale. Or use target weight as lower bound.

Also, in a year chart there's no need to reserve the scale up to end of the year. Doing so in January we will get almost blank screen. X-line should have 1st of January as 0 and the current date (plus some margin, say, one week) as its right bound.

And the last, putting all the values on a chart isn't a good idea when you have many records. Better not to put at all (or have a setting to put or not) or put only for peak values.

Look what I have in openScale and other similar program with exactly the same data:

[image]

[image]

[image]

alekksander, 2017-03-03:
good point

oliexdev, 2017-03-03:
Hi taraska7

thanks for your comments. Let me response directly to your comments.

> I'd like to mention that the scale you use while building charts (starting from 0) isn't the best solution. This results in almost straight line, especially on small screen of a smartphone. This [...]

It is already exactly implemented like you described. The margin is +/- 4kg. ;)
The problem is that your graph has zero values. The y-axis range have to be large to show these values. Try to disable all measurements in the settings and you can see the y-axis is adjusted. But you are right its better to decompose the y-axis of the weight and the other measurements. I think I will introduce a y-axis on the right side of the graph for circumferences. What do you think?

> Also, in a year chart there's no need to reserve the scale up to end of the year. Doing so in January we will get almost blank screen. X-line should have 1st of January as 0 and the current date (plus some margin, say, one week) as its right bound.

The space is not reserve up to the end of a year. In the graph you see only a month, so maximum there are about 30 days left. I think its ok.

> And the last, putting all the values on a chart isn't a good idea when you have many records. Better not to put at all (or have a setting to put or not) or put only for peak values.

There is already a setting where you can disable/enable the labels on the graph. But maybe I can add an option to individual configure the data point graphics (e.g. dot, square, diamond or none) 

Best regards
Oli

taraska7, 2017-03-03:
> It is already exactly implemented like you described. The margin is +/- 4kg. ;)
The problem is that your graph has zero values. The y-axis range have to be large to show these values. Try to disable all measurements in the settings and you can see the y-axis is adjusted. But you are right its better to decompose the y-axis of the weight and the other measurements. I think I will introduce a y-axis on the right side of the graph for circumferences. What do you think?

Ok, got it =) I had waist circumference that I haven't measured yet while weight is ported from other program. Obviously, different measurements have to be decomposed. They even have different units (kilos for weight and cms for waist circumference). They may be seen on the same screen, but every chart must use its own y-line values.

> The space is not reserve up to the end of a year. In the graph you see only a month, so maximum there are about 30 days left. I think its ok.

My fault, I was confused because there's 2017 on the top. Anyway, I'd like to see my all-year progress, not only a given month. Could you please explain what is showed on the lower chart? As far as I understand, this is the number of records. Quite useless, doesn't it?
[image]

> There is already a setting where you can disable/enable the labels on the graph. But maybe I can add an option to individual configure the data point graphics (e.g. dot, square, diamond or none)

I'd like to ask you to put this on todo list. By the way, I found why I was confused again with these labels. When I change this setting or change the measurements I'd like to have, it is necessary to fully stop the app and start it again. It's easy (but still annoying) on my Cyanogenmod, but could be tricky on a stock Andriod for the average user. And no warning about it in settings.

Have a nice weekend!

oliexdev, 2017-03-04:
> My fault, I was confused because there's 2017 on the top. Anyway, I'd like to see my all-year progress, not only a given month. Could you please explain what is showed on the lower chart? As far as I understand, this is the number of records. Quite useless, doesn't it?

On the lower chart are the months of a year. The height of the diagram represent the amount of measurements. I don't think its useless because you can clearly see how many measurements you have done in a month. Even more you don't have to show the complete year just the month in detail. 

> I'd like to ask you to put this on todo list. By the way, I found why I was confused again with these labels. When I change this setting or change the measurements I'd like to have, it is necessary [...]

I implemented an option to hide or show the dots of graph, see commit https://github.com/oliexdev/openScale/commit/b7486f66c00f22f24c9bfe1fe3dcb6954311d11b
You are right its a known bug that a setting change doesn't sometimes not immediately update the data. I will open a new issue on that. Need to figure it out someday how to fix that.

I will do the y-axis decomposing together with the issue https://github.com/oliexdev/openScale/issues/4

closed 2017-03-05 by oliexdev