oliexdev/openScale #9: confirm to delete

alekksander, 2017-03-03:
we don't want our precious data to be deleted by accidental touch

oliexdev, 2017-03-03:
Hi alekksander

thanks again for your comment. You are right that shouldn't be possible but while I were implementing the delete function I thought it could be also useful if you have lot of messed up weight data to delete, so you are not annoyed to confirm every deletion. 
I could add an option like "don't ask me again" in the confirmation dialog or I could add a function to restore a deleted measurement? 

Best regards
Oli

alekksander, 2017-03-03:
Hello. Thank You for response. 
"don't ask me again" seem like the adequate solution to me, but the other one is fine too.

oliexdev, 2017-03-04:
I implemented a delete confirmation dialog which you can enable or disable in the settings, see commit https://github.com/oliexdev/openScale/commit/d35bcca8d609b926d2c60f3f87736738e031daa8

closed 2017-03-05 by oliexdev