oliexdev/openScale #12: Bluetooth Connection

spdizzle, 2017-03-07:
Hey, first of all great app, thanks for your service. 
I have a problem connecting the MI_SCALE with Bluetooth. The openScale message (no device found).
The device is found in the Android Bluetooth Management Tool. 
Connection works with the official MI Fit App.
Is there a special pairing procedure?

Thx

oliexdev, 2017-03-07:
Hi spdizzle,

thanks for your question. No there is no special pairing procedure. If you first connect to the scale or after a battery change there is only an initialization procedure, see [openScale wiki](https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale) for further information.

Here are some ideas why it might not work:

- do you already connected to the scale via bluetooth while are you trying to connect with openScale? Because if you already connected you can't establish a new connection.
- Did you already try a few times consecutively to connect to the mi scale? 
- Did you change the device name of the mi scale? The default name is `MI_SCALE`. You can set the device name in the settings but be aware that the device name is case sensitive.
- You can install [BLE Scanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner) and search for the device if you can connect to the scale, please check the bluetooth device address. It should start with `0x880f10`. 
- Can you tell me the software version of the bluetooth scale? (It should be `v0.5.8`)

I hope this will help you.

Best regards
Oli

spdizzle, 2017-03-07:
Hey Oli, thanks for your fast answer. I've uninstalled the mi fit app and changed the batteries. I tried to connect a few times, the device name should be allright, software version is v0.5.8).

Maybe I'm too stupid for the initialization procedure. I have installed the BLE Scanner, i can connect to the MI_SCALE but the scale is not bonded. I turned on the notification flag, added the value, but then i had a disconnect. It would be very nice, if you can give me a step-by-step tutorial for the initialization :)

oliexdev, 2017-03-07:
The initialization process shouldn't be the problem. If you install the mi fit app and start it once, it should be initialized by the mit fit app otherwise openScale will do it then. 
You can verify it using the BLE Scanner go to `Weight Scale -> Current Time` and read out the hex value, It should start with `0xE107` (little endian format for year 2017)
What makes me wonder is that you can't connect to the scale at all with openScale. You don't have to be bonded. Did you verify the `0x880f10` device address? You can see that in the overview in the BLE Scanner. 
Which notification flag did you enable?

spdizzle, 2017-03-07:
Can't find the 0x880f10 value, the value at device information is 0x180A like in your wiki. At the overview the value is C8:0F:10:A1:8A:E0. 

the uuid value at current time is: 00002A2B...
I've enabled the flag at weight scale --> custom characteristics, but i didn't work.

oliexdev, 2017-03-07:
ok can you read out the hex value under `device information -> system id` or make a screenshot of the BLE Scanner overview.
The UUID is correct but you need to check the hex value. Click on the R Button next to the `Current Time` then under hex a value is pop up.

spdizzle, 2017-03-07:
The hex value starts with 0xC80F10, maybe thats the error, how can i change that to '88'

oliexdev, 2017-03-07:
Interesting that your device address starts with a different hex value. That should be the problem. You can't change that but I updated the openScale source code so that the device address is not checked anymore, see commit https://github.com/oliexdev/openScale/commit/54181116a5e0c59a1a6ec8e07f385ffc39470c48
So in the next version of openScale it should work. If you don't want to wait you can compile the latest source code by yourself with for example Android Studio.

closed 2017-03-10 by oliexdev