oliexdev/openScale #13: no bluetooth device found

TjrGithub, 2017-03-09:
(possibly a dupe of #12)
Hi, I bought the scale because I saw your app on FDroid. Thanks for all your hard work. Unfortunately, I have trouble connecting it:

(1) Maybe there could be a paragraph in the about screen how to connect to the scale? Even if the process is dead simple, laying out the steps removes a source of confusion.

I managed to pair it this way: put phone on scale with bluetooth enabled, re-insert batteries, try to pair from phone bluetooth settings. This gets lots of error messages "Couldn't pair with MI_SCALE because of an incorrect PIN or passkey." but finally it worked.

(2) The app cannot find the scale even if the phone and the scale are connected (and also if they are not). Instead, I get a toast message about no bluetooth devices found.
All the bluetooth info of the scale is here (only for one week for privacy reasons): http://pastebin.com/zne8xUd3
Bluetooth info was captured using https://f-droid.org/repository/browse/?fdfilter=blexplorer&fdid=org.ligi.blexplorer

Phone OS: Android 6.0.1 Slimroms

oliexdev, 2017-03-10:
Hi TjrGithub,

yes you have the same problem like #12 
I hope I figured it out now. The problem is that you have a different mi scale hardware address. Your address starts with `C8:OF:10` rather than `88:0F:10` but I won't check the address any more in the new release. So please upgrade the openScale version to 1.4.3 and try again. 

Best regards
Oli

closed 2017-03-10 by oliexdev

TjrGithub, 2017-03-10:
Ok, thanks for taking this seriously and even fixing this so fast. I created tickets for the remaining two subissues, i.e. missing pairing instructions and upgrade availability. I'll test with 1.4.3 as soon as it's available and report back.

TjrGithub, 2017-03-10:
Confirmed working on Android Slimrom 6.0.1 and Blackberry 10. Except for one minor hiccup: When the Blackberry has finished reading from the scale, I need to turn off bluetooth on the blackberry and let the scale sleep before the Android will connect. Else it just says no devices found.