oliexdev/openScale #14: missing instructions how to connect phone <> bluetooth

TjrGithub, 2017-03-10:
(1) Maybe there could be a paragraph in the about screen how to connect to the scale? Even if the process is dead simple, laying out the steps removes a source of confusion.

I managed to pair it this way: put phone on scale with bluetooth enabled, re-insert batteries, try to pair from phone bluetooth settings. This gets lots of error messages "Couldn't pair with MI_SCALE because of an incorrect PIN or passkey." but finally it worked.

oliexdev, 2017-03-10:
I don't have an about screen ;) and there is no pairing process just click on the bluetooth symbol and that its. Just make sure you have the right device name under the settings.
What you tried is to pair your bluetooth device with your mobile phone but you don't have to do it.

TjrGithub, 2017-03-10:
That's good to know. What you write here would have saved me some fiddling around.

How about putting this information into the preferences, bluetooth tab, an (non-interactive) fifth menu item right under smart user?

oliexdev, 2017-03-11:
If you don't know how you do it you can first search here for an answer otherwise create a new issue. But usually the app should be self explained.

closed 2017-03-11 by oliexdev