oliexdev/openScale #16: Support Bluetooth LE weight scale profile

JavaJens, 2017-03-10:
Hi, I have a Polar scale which should broadcast its data according to the BLE specs. 
It would be cool if the app can support the weight scale profile, so that a multitude of scales can easily be supported. 
Keep up the great work!

oliexdev, 2017-03-11:
Hi JavaJens,

first of all thanks for your compliment.
Unfortunately many companies don't adhere to the official weight service of the ble specification. As I don't own a polar scale I can't even take a look and don't even test it. But if you are a developer and have the effort to reverse engineer your polar scale feel free to create a pull request. A first hint how you can do it is the [openScale wiki](https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale) page.
I can't help much with the reverse engineering stuff because its a lot of trial and error but if you have question how you implemented it in openScale I can help you of course.

Best regards
Oli

closed 2017-03-13 by oliexdev

dhx256, 2020-03-15:
Hi
I have a Polar scale. Captured some traffic and run the discover-routine in openscale.

In the logs are 5 measurements:

93,5 (3A7) line 44303
99,6 (3E4) line 44479
109,2 (444) line 44769
9,8 (0062) line 44926
15,7 (009d) line 45081

hope someone want to do some work with it.

[openScale_2020-03-15_11-46.txt](https://github.com/oliexdev/openScale/files/4334280/openScale_2020-03-15_11-46.txt)

[btsnoop_hci_202003151410.zip](https://github.com/oliexdev/openScale/files/4334284/btsnoop_hci_202003151410.zip)

bmmmm, 2022-07-11:
Hey hey, do we want to pick up this topic again?
I am able to contribute some data, if needed.
I have a Polar balance Model: G1 available.

Looking forward to include Polar Balance to the project!
-Bartosz