oliexdev/openScale #17: No Bluetooth device found.

Sarkasper, 2017-03-13:
I tried a couple times to connect but it won't work. 
If you need more info let me know.

[image]
[image]
[image]

oliexdev, 2017-03-13:
Hi Sarkasper,

thanks for your hint. Its seems like devices with starting hardware address `C8:OF:10` makes trouble. Which openScale version do you use and what is your SmartPhone and Android version?

Best regards
Oli

Sarkasper, 2017-03-13:
[image]
[image]

Sarkasper, 2017-03-13:
I found the new release of version 1.4.3 and that has fixed it, thanks!

closed 2017-03-13 by Sarkasper