oliexdev/openScale #20: App sometimes crashes when trying to connect to Mi Scale

Foxdelmar, 2017-03-29:
Hi oliexdex,

its great to see someone creates open source apps that save your privacy for products that want to share everything with the cloud. Thank you.

I bought the Mi Scale, downloaded your app from F-Droid with a LG G4 with Cyanogenmod 13 (Android 6.0.1) and tried to connect it to the scale. I also deactived the privacy features for this app. Only the firewall (Afwall) is activated.

Often it works and I can receive the values from the scale. But more often the app crashes. Sometimes immediately after the start and sometimes after I tap on the bluetooth Symbol within the app.

Then it works for quite a few times and after a while it is broken again --> crash after trying to connect.

Maybe there is a problem if I start the app when I am standing on the scale already and the scale is still changing the weight quickly.

I can not locate the problem in detail but maybe you experienced similar things. 

How does the app handle BLE-information that it does not understand? Does it ignore it or might this be the reason for the crash? Maybe you could catch errors within the app?

Best regards
Fox

oliexdev, 2017-03-30:
Hi foxdelmar,

thanks for your comment. Can you quick check if you see in settings-> Bluetooth-> device type `xiaomi mi scale v1` has in the end `v1`?

Best regards 
Oli

Foxdelmar, 2017-04-04:
no. the device type says
"Gerätetyp: Xiaomi Mi Scale"
"Gerätename: MI_SCALE"

Should I change this?

Best regards
Fox

oliexdev, 2017-04-05:
As I guessed, I already fixed this bug but FDroid didn't update it. Please download the lastest version on https://github.com/oliexdev/openScale/releases or install it over Google Play.

closed 2017-04-05 by oliexdev