oliexdev/openScale #24: Integration with Google Fit

Larry0ua, 2017-04-10:
Hi there,

Do you have any plans for integrating with other services? I am interested in Google Fit integration - there is a possibility to transfer body weight data to that service. Now I cannot find an app that supports this integration easily... What I'm asking now is your attitude about this - I may develop a PR with this integration, but some steps will be required from the application owner to allow this integration

oliexdev, 2017-04-11:
Hi Larry0ua,

No currently I don't have any plans to integrate any services. I am not a fan of cloud service for health data for privacy reasons. Can I ask why do you want such an integration in Google Fit? Why don't you enter your weight not directly via Google Fit if you want to use it anyway?

Best regards
oli

Larry0ua, 2017-04-11:
That's a shame to have wireless scales and enter data manually :) I didn't find any application that allows this integration yet.

oliexdev, 2017-04-12:
maybe you can export and import it? I don't know never used Google Fit...

closed 2017-04-12 by oliexdev