oliexdev/openScale #25: Translation of description text into French

oliexdev, 2017-04-12:
It would be great if somebody could translate the following description text into French:

> openScale is an Open Source app to keep a log of your weight, body fat, water percentage and muscle percentage.
> 
>  **Features**
>   - Logs your weight, body fat, water percentage and muscle percentage
>    - Display all your data on a chart and table
>    - Import or export your data from/into a CSV file
>    - Support for multiple users
>    - Support for Bluetooth Xiaomi Mi Scale
>    - Connection to a custom made bluetooth scale (see openScale wiki for more information)
>    - All data belongs to you (no cloud service)

Thanks to scritesair the app itself is already translated to French see PR #21 

Best regards
oli

scritesair, 2017-04-12:
Hi Oli,

The french transation I submitted was the first commit I made on Github :)
I didn't see your answer, but no problem for the translation of the description.
Here it is!

------------------------------------------

openScale est une application open source qui vous permet d'enregistrer votre poids, votre graisse corporelle, votre pourcentage d'eau et votre pourcentage de muscle.

**Fonctionnalités**

- Enregistrer votre poids, votre graisse corporelle, votre pourcentage d'eau et votre pourcentage de muscle

- Afficher toutes vos données sur un graphique et un tableau

- Importer ou exporter vos données depuis/vers un fichier CSV

- Prise en charge de plusieurs utilisateurs

- Support de la Xiaomi Mi Scale via bluetooth

- Connexion à une balance bluetooth personnalisée (voir le wiki openScale pour plus d'informations)

- Toutes les données vous appartiennent (aucun service cloud)

--------------------------

Thank you for this awesome app!

Greetings from France
Scritesair

oliexdev, 2017-04-13:
merci :)

Unfortunately for the short description on Google Play the French translation is too long. Google Play allows only 80 characters. Is it possible to shorten this sentence in French?

> Keep log of your weight, body fat, water percentage and muscle percentage.

Google Translate says:

> Gardez le journal de votre poids, graisse corporelle, pourcentage d'eau et pourcentage de muscle.

But this is still too long (97/80 characters)...

Greetings from Germany
Oli

scritesair, 2017-04-13:
Hi!

First, I was very happy to see my translation on F-Droid on the 1.5.0!
I didn't know that Google Play allows only 80 characters for the description.

The translation of Google translate is unfortunately incorrect.

I think you can cut here :

> Enregistrer votre poids, graisse corporelle, pourcentage d'eau et de muscle

(79 characters \o/ )

which corresponds to : 

> Logs your weight, body fat, water and muscle percentage

instead of (litteraly) :

> Logs your weight, your body fat, your water percentage and your muscle percentage

Keep me informed if there's any other problem!

oliexdev, 2017-04-13:
that was a fast response thanks scritesair now the sentence fits in.
In a few hours you will see your translation on French Google Play ;)

Happy holidays
oli

closed 2017-04-13 by oliexdev