oliexdev/openScale #27: multi-year view in chart

c64bob, 2017-04-24:
First, thanks for a great app! I've been using it for a few days and really love it.

In the charts tab, would it be possiblt to implement a yearly view, as well as a view with all available datapoints? I'd like to be able to see my total progress, as well as the progress over a longer period than a month.

Thanks again.

oliexdev, 2017-04-24:
Hi c64bob,

thanks for your comment. I am not sure how to best implement a yearly view without breaking the design concept with the months!? Currently you can view the total process of **all** your data on the table view fragment.
But what kind of datapoints are missing on the chart? Finally let me say a little remark if you want to show all your data in **one** chart you will have the problem of different x/y axis scales.

Best regards
oli

c64bob, 2017-04-25:
Hi Oli,

thanks for your quick reply. I used to use the application [droid weight](https://code.google.com/archive/p/droidweight/) to manage my measurements. The code hasn't been updated in a long time, and with the recent upgrade to Nougat I can't use the app properly anymore. It did have a chart view for various time frames, including all available data. To avoid overload of too many data points on the chart, it might have averaged some of the points if neccessary. Maybe there are libraries available the offer this functionality?

oliexdev, 2017-04-25:
Do we talk only about weight datapoints? Because I assumed you mean with "all datapoints" the other measurement too like waist/hip circumference, water/muscle/fat percentage and so on.
I am using the [hello-chart library](https://github.com/lecho/hellocharts-android). It don't have a function to average the datapoints. Maybe I could add a button to show/hide the lower month diagram and if its in hide mode automatically a year is shown?!

c64bob, 2017-04-26:
You're right, I was only refering to the weight datapoints. I'd like to see all available weight measurements in one chart to check on my progress, as well as identify any longer term trends. May averaging isn't even needed, if the library handles the display of lots of data on one chart automatically? If it is easier for you, leave out the yearly view, and just implement a total view without the lower month diagram?

Thanks again for your efforts, greatly appreciated!

oliexdev, 2017-04-29:
I implemented a button to show or hide a yearly diagram overview with commit https://github.com/oliexdev/openScale/commit/cc60815d75ee3ff14fe2928f8e93d2056d560509

closed 2017-04-29 by oliexdev