oliexdev/openScale #28: script to import database from libra

feclare, 2017-04-25:
I created a simple python script to transform the csv file from libra to the openscale csv format. 
https://gist.github.com/feclare/9ce446a640ac1b8854456a4da468dc05

oliexdev, 2017-04-25:
Hi feclare,

awesome thanks for the script.

Best regards
oli

closed 2017-04-25 by oliexdev

sa5bke, 2023-09-20:
I googled importing from libra and came to this issue. The mentioned script is for python2 so I had to do some small changes to get it work in python3:

```diff
10c10
<     print("Missing file to transform\n")
---
>     print "Missing file to transform\n"
13,14c13,14
< r = csv.reader(open(sys.argv[1]), delimiter=";")
< writer = csv.writer(open("openScale_data_Libra.csv", "w"), delimiter=",")
---
> r = csv.reader(file(sys.argv[1]), delimiter=";")
> writer = csv.writer(file("openScale_data_Libra.csv", "w"), delimiter=",")
```