oliexdev/openScale #29: BMI calculator units

maxb2, 2017-04-25:
When I use cm for height and lbs for weight, the BMI is calculated incorrectly. It looks like the weight in lbs isn't being converted to kg before the BMI calculation. The same thing is happening with weights in st.

oliexdev, 2017-04-25:
Hi maxb2,

that is true, its a bug. Is the BMI calculation for st or lbs always converted into kg or is there a different calculation possible?

Best regards
oli

maxb2, 2017-04-25:
[BMI](https://en.wikipedia.org/wiki/Body_mass_index) (body mass/height<sup>2</sup>) is always defined in units of kg/m<sup>2</sup>. Converting between kg, lbs, and st is easy though. 
1 lb = 0.454 kg
1 st = 6.350 kg
1 st = 14 lbs

On the same subject, it would be nice if previous entries would be converted automatically after changing which units you want to use in the settings. Maybe you could store everything in kg and then convert the numbers when it needs to be displayed. Or it may be easiest to just store all three numbers.

oliexdev, 2017-04-29:
fixed it with commit https://github.com/oliexdev/openScale/commit/687c0b95ca32a5561b269ea5b7ca2fcaecd4eb9a
weight data points are also automatically converted after changing the unit.
But be aware that the exported data is always in kg!

closed 2017-04-29 by oliexdev