oliexdev/openScale #31: Add measurements in 1-step again

patzzz, 2017-04-30:
In 1.4.3 version I could add all measurements in 1-step and old values could be edited in the same screen, for example change only decimals:
[image]

But in 1.5 version I need 5-steps and old values aren't editable.

First step to add weight:
[image]

Second step to view old values:
[image]

Third step to add water percentage (you can't change old value):
[image]

Fourth step to add muscle percentage (you can't change old value):
[image]

Fifth step to add body fat percentage (you can't change old value):
[image]

Is possible change it again to 1-step changing old values?

Thank you very much.

oliexdev, 2017-04-30:
Thanks for your comment. That is true that you can't change easily the position after decimal point, so I will show the old value in the edit text with the commit https://github.com/oliexdev/openScale/commit/b7befd8aa8837375d29e93bd69cdd7fa2da1b0ef . But I am not quite sure if it's better or worse to have an edit text instead of an input dialog.

patzzz, 2017-05-01:
The 1-step screen of 1.4.3 version was perfect for me, but perhaps a setting to add measurements in old-style or new-style (with multiple screens and an empty text dialog) would be perfect for all.

Thank you for this fantastic app.

oliexdev, 2017-05-03:
let me think about that which method is better but I don't want to have two input modes.

closed 2017-05-03 by oliexdev