oliexdev/openScale #36: Wrong data (lbs/kg) after updating

mibanescu, 2017-05-08:
After updating to openScale 1.5.1, my data was all wrong.

My user's preferences were set in lbs (instead of kilos). After the update, the user still had the preferences in lbs, but the app was showing all old values 2.2 times larger (i.e. I think it interpreted them as kilos).

In order to revover, I had to export the data, modify it in Excel (it was still showing up in lbs, so I had to convert it to kilos), clear the data and cache for openScale, re-create the user with the preference in kilos, import the data, then change the preferences to lbs again.

oliexdev, 2017-05-08:
Hi mibanescu,

thanks for your hint. I didn't consider such situation. 
In version 1.5.1 I calculate and store internally all weights in kg and just convert to lbs/st to fix BMI issue #29.
A side effect is that you can now change easily between the units. Now all weight data are automatically recalculated according to the selected unit.
But the drawback is that values in lbs or st that are used before version 1.5.1 has to be manually  converted into kg like you described. Sorry for the trouble.

Best regards
oli

mibanescu, 2017-05-08:
No problem at all, I was happy I was able to get my data back.

May I suggest saving the unit(s) of measure as part of the export (and reading them as part of the import)? It would be helpful if one mis-configures the user before importing for the first time.

Another possibility would be for the export to include the user's preferences (name, goal width, units of measure etc) for easier transfer of that file between devices. As long as the export follows the units of measure from the backup, the import could make decisions on what the numbers mean.

Saving the user info in the same csv file would make things a bit more sketchy, you'd have to add headers to separate the data. But it would make for a much more robust import/export if in the future you need to add a column.

Just suggestions.

oliexdev, 2017-05-10:
thanks for your suggestions if I have time I will try to implement that.

mibanescu, 2017-05-11:
Frankly, I don't like my own suggestion of putting the user dump and the measurement data in a single csv file. I think they make sense to be together, but when I think of csv I think of a single "table" (in sql sense). yaml would make more sense, but yaml is also more verbose and harder to manipulate with other tools like OpenOffice/Excel/etc. Maybe having a separate export/import for user information would be the happy medium.

oliexdev, 2017-05-11:
Well I guess to export the weight to the selected user unit would be not bad. 
I don't think it is really necessary to export the user information separately because it hasn't got much data and you can create easily a new user.
But I might consider to implement a method to 1:1 copy the internal databases for backup and transfer purpose because the database has also some more internal information that are not exported.

mibanescu, 2017-05-11:
You may want to look at CarReport (also open source, for tracking car expenses like gas, repairs etc - I use it mostly to track my fuel efficiency). https://bitbucket.org/frigus02/car-report

It has backup support to Google Drive and Dropbox, I believe it's sqlite behind the scenes, and the database itself is easy to transport.

oliexdev, 2017-05-11:
thx but I don't like to upload any health data to a cloud service but of course If the user want that they can save the backup into the Dropbox or Google Drive folder (I will make the data path editable). 

Yes Android use sqlite it also really simply to make a backup just raw copy the database to another directory.

mibanescu, 2017-05-11:
I totally understand about the privacy concern.

oliexdev, 2017-05-23:
implement a basic database backup function with commit https://github.com/oliexdev/openScale/commit/f02130a61ef9453b463b298bcebdcbb6a32cb5de

closed 2017-05-23 by oliexdev