oliexdev/openScale #37: wrong dates and times (Mi Scale firmware v0.5.8)

momo42, 2017-05-09:
Hello oliexdev,

first I want to thank you for this great app. Me and my wife also don't like to push our personal data to the cloud. When I found your app, I decided to order a Mi Scale for us.

But now we have headaches with the dates of the measurements. Me and my wife use two phones with openScale (1.5) to track our weight. On both phones we first created two accounts in openScale. Then we did some test measurements and synced our phones to the scale. We had to delete some of the first values on each phone because they went to the wrong user, but despite of that everything went fine.

But when we used the scale again two days later, all new synced values got dates and times of our test measurements, or maybe the date of an deleted test measurement. So we took out the batteries of the scale and reinstalled openScale and took some test measurements but the next day the same thing happened. All new measurements get the date and of one of the test measurements. Deleting some or all measurements in openScale also didn't help.

I mentioned the fw version because I think that this could be the root of our problems. When we got the scale, I first connected it to Mi Fit and after a break of a minute or two Mi Fit displayed the firmware of the scale as "v0.5.8". Mi Fit didn't mention any update process but perhaps it had silently updated the scales firmware. I think so, because when you connect a Mi Band it gets updated as well. And new fw versions often lead to problems with Gadgetbridge, an open source app for mi band sync (see https://github.com/Freeyourgadget/Gadgetbridge/wiki/Mi-Band-Firmware-Information).

Could anyone please try to sync his Scale to Mi Fit and report back, what firmware version is displayed? And then check again, if syncing to openScale shows the right dates for new measurements. Perhaps Xiaomi changed the transmission protocol ...

And another question:
- How many measurements are stored in the scale and transmitted as history when you sync? And are the transmitted measurements deleted after that? (the openScale wiki is not clear about that) I ask because we want to use our two phones alternating and don't want to miss weights on each of them. And we don't want to take the "live" weight holding our phones in the bathroom. Instead we would like to use the history to sync our measurements every day or two. Is this possible?

regards
Momo

please excuse my bad English!

oliexdev, 2017-05-10:
Hi Momo,

thanks for your comment and your compliment. 

I seemed like your mi scale is not correct initialized.
First idea and mostly the save way is to start once with the original Mi Fit App after you remove and reinsert the batteries. The Mi Fit will initialize the scale.
Second is to initialize your scale with openScale (I tried to reverse engineer the initialization process). Remove also the battery and insert it again. Then when you trying to connect a short notice `Initialize Bluetooth device` should pop up.
The hw version of the mi scale is `v0.5.8` 
Regarding to your last question about the deletion of transmitted measurments. The openScale Wiki says:

> The last think that makes me wonder is that you can't delete the history weight measurement on the Mi Scale at all or I didn't find it out yet.

So no there is no way (or I didn't find it out yet) that any measurements are deleted on the scale! So if you are using two phones the best way is to create on both phones two users to separate the incoming data according to the weight by using smart assignment. (Note that you have to manually add the first weight of each user, so openScale knows which weight assign to the correct user)
If you complete delete all data in openScale (for example reinstall openScale) you will receive every time all your test measurements and history measurements that you have ever done with the mi scale.

Best regards
oli

kapitaniglo11, 2017-05-14:
Hello guys!

I don't know that it's the best place, but recently I have problems with synchronizing weight using openScale on my Mi Scale as well.

I weight myself everyday, in the morning. When I use only openScale to sync my results, they have weird dates/times. Like all of them on the same day, just different times across the day (morning->evening). It seems to fix it, I can launch Mi Fit app, sync in Mi Fit (the results there are normal), and then sync in openScale again, but then the results are duplicated. I still have those weird "all on one day" + normal ones with the same measurements but correct times.

Android 7.0, newest openScale (1.5.1) , newest Mi Fit (2.4.2) + scale firmware v.0.5.8

P.S.
I really love your app, and I think it's not getting credit it deserves.

Kind regards,
KapitanIglo

momo42, 2017-05-14:
Your app indeed is a very nice piece of software. It shows, that all the cloud foo is not necessary for functional fitness apps, and that opensource apps can look pretty too. So keep on the good work! I'll try to help you sorting out this weird date bug.

I did some (sadly not very systematic) testing over the last two weeks (just with one phone and just Mi Fit OR OpenScale 1.5) and can partly confirm the same behavior than Kapitaniglo. My OS is LineageOS 14.1 (Android 7.1). I don't remember the Mi Fit version.

I used Mi Fit (only) for some days and the dates in Mi Fit are correct.
Then I took out the scale's batteries, synced the scale to Mi Fit again, installed OpenScale and then deleted the Mi Fit app to avoid any troubles between Mi Fit and OpenScale. So I did the initialization with Mi Fit. When I used OpenScale for some days after that, all the synced measurements had different times but the same date.

I'm pretty sure I also did the initialization (scale's barries out ...) after that with only OpenScale installed and got the same results. I never tried OpenScale and Mi Fit alternating.

I would really like to help, but it is hard for me to do a systematic test sequence over three ore more days. My wife is already complaining about the new 'cyber' scale not being available in the morning, advantages of old fashioned mechanical scales etc. :-D 

And one more question about the measurement history. Did I get you right, that every time you sync, ALL the old values are synced between she scale and the phone (every day some values more than the last day, perhaps hundreds of values ... since the batteries die some day)? How does your delete function work, if the same (in the app deleted) values get synced every and every day again? Please excuse my dump questions. I'm not a programmer. ;-)

kapitaniglo11, 2017-05-15:
> If you measuring your weight with openScale your date is always the same but the time is different?

Yes. Then I would open Mi Fit App...
> If you measuring with the Mi Fit app you date and time is correct?

Yes. Everything there would be correct.

> If you using openScale again, you get the wrong weights and the correct weights?

Yes. There are still the "same day" ones + the corrected ones that I saw in Mi Fit. 

Downloading your beta app and will give you feedback after few days!

Hear you then!

kapitaniglo11, 2017-05-16:
Using just updated "beta" openScale app, without Mi Fit.

First log, and already wrong time. Around 9:20 described as 20:19 by openScale. Pic related. 

[image]

kapitaniglo11, 2017-05-16:
Another update after 2nd usage of the day.

I weight myself again, turned on the app, and it synced - I have "fake times" and "real times" now again... Everything is duplicated. Pic related

[image]

oliexdev, 2017-05-16:
that is really strange you weight once but get two measurements? one is right the other one is wrong? So in the first picture the measurement at 15.05.2017 20:19 is wrong but you see the correct measurement in the second picture at 16.05.2017 9:19.

Did you get the two measurements at 16.05.2017 9:32 and 16.05.2017 14:36 at the same time while you weighting only one time? 

Are you able to use Android Studio to test the source code? Because we need the raw hex values of the scale. Would be more easier if you could post the log output of the app.

kapitaniglo11, 2017-05-16:
Yes, yes, and yes. Both of them when syncing one time. 

Unfortunately, I am not able to use Android Studio. Opened GitHub account just to contact you.

It *seem*, that it can work correctly, when I would use Mi Fit first, and then just open openScale to fetch the "somehow changed by MiFit" data... I can try it next.

oliexdev, 2017-05-18:
>Yes, yes, and yes. Both of them when syncing one time.

No not when syncing one time when you **weighting** one time then you get two measurements at once?

> Unfortunately, I am not able to use Android Studio. Opened GitHub account just to contact you.

Ok I should consider to implement a debug mode for the official version... that would be useful situation like that.

>It seem, that it can work correctly, when I would use Mi Fit first, and then just open openScale to fetch the "somehow changed by MiFit" data... I can try it next.

It is curious that you don't weight again but Mi Fit add an extra measurement to the scale base of the incorrect measurement!? That is totally strange...

> Tested it. After syncing with MiFit (everything tip-top), and then opening openScale, it imports just one, correct data.

I am confused why it only sync one measurement now?

Can you describe step by step when the correct measurement is visible. When do you syncing with openScale, in which step do you weight and on which step do you syncing with Mi Fit.

kapitaniglo11, 2017-05-18:
It is confusing for me as well :D Mi Fit is not adding extra measurements. There you have got only proper results.

Anyway, step by step.

1. Weigh myself
2. Go to the mobile in the living room.
3. Open openScale.
4. Sync openScale.
5. Getting bad result + normal result at once. From one weighting in.
6. Weigh myself again
7. Go to mobile
8. Open MiFit
9. Everything there is OK. The previous one is with the correct time, and the most recent is with the correct time
10. Open openScale to see how it would fetch data now.
11. After pressing sync it syncs the most recent one just with the correct time now. 
12. In OpenScale I am left with (fake+real,real), and in MiFit I have (real,real).
13. In other words - I can not use *just* openScale. I need to use MiFit to sync, and then sync again with openScale. It is the only way to get correct results.

oliexdev, 2017-05-18:
Thanks for your detailed step by step analysis. I think we getting closer to the problem.

Can you test two another step orders?

1. Can you do step 1 to 5 two times in a row?
2. Can  you do step 6 to 9 first and then do step 1 to 5

What are the results of these two step orders?

kapitaniglo11, 2017-05-18:
I must admit that I messed up my first description. 

There is *REVISED* 5. Getting bad result, so I am running Mi Fit to check how it would behave, it displays correct one, so I am running openSync again, and getting now normal result as well. I am left with normal+wrong in openSync.

I did what you asked for.

1. Weigh myself (69.0)
2. Open & sync openScale.
3. Getting bad result (11:06).
4. Open MiFit, see correct time 22:51.
5. Open & sync openScale again. It adds 22:51 now.
1a. Weigh myself again (69.0)
2a. Open & sync openScale.
3a. Getting bad result (11:07).
4a. Open MiFit, see correct time 22:53.
5a. Open & sync openScale again. It adds 22:53 now.
6. Weigh myself again (69.5), open Mi Fit. 
7. See correct time (22:56).
8. Open openScale, sync.
9. Getting just correct one (22:56)
1b. Weigh myself again (69.7)
2b. Open & sync openScale.
3b. Getting bad result (11:08).
4b. Open MiFit, see correct time 22:57.
5b Open & sync openScale again. It adds 22:57 now.

And I am left with what's attached:
[image]

oliexdev, 2017-05-19:
> There is REVISED 5. Getting bad result, so I am running Mi Fit to check how it would behave, it displays correct one, so I am running openSync again, and getting now normal result as well. I am left with normal+wrong in openSync.

That's bad I thought you get at least once a correct value from openScale.

> I did what you asked for......

Thx it seems like Mi Fit manipulate the date/time on the scale when you open the Mi Fit app :/
That behaviour is totally strange...

We need to see what is send between the mi scale and openScale/MiFit app

Can you turn on the developer options of your smartphone and follow these procedure:

1. Turn on the `Bluetooth HCI Snoop Log` option on your smartphone 
2. Weight yourself and write down on a paper the correct date and time
3. Open openScale and sync it with your scale
4. Close openScale
5. Turn off the `Bluetooth HCI Snoop Log` option on your smartphone 
6. Save /sdcard/btsnoop_hci.log to another location and rename it to "sync_openscale_hci.log"
7. Delete /sdcard/btsnoop_hci.log
8. Turn on again the `Bluetooth HCI Snoop Log` option on your smartphone 
9. Open Mi Fit app and sync it with your scale
10. Close Mit Fit app
11. Turn off the `Bluetooth HCI Snoop Log` option on your smartphone 
12. Save /sdcard/btsnoop_hci.log to another location and rename it to "sync_mifit_hci.log"
13. Post both logs here and write down which date and time was displayed in openScale and MiFit in step 3 and step 9

In this way I can analyse the bluetooth traffic with wireshark between the scale and openScale/MiFit app.

kapitaniglo11, 2017-05-19:
Here are the logs: [MiScaleLogs.zip](https://github.com/oliexdev/openScale/files/1014852/MiScaleLogs.zip)

Weight was: 70.0kg @ 16:59, 19th May 2017
openScale result:  70.0kg @ 9:00, 19th May 2017
MiFit result:  70.0kg @ 16:59, 19th May 2017

oliexdev, 2017-05-19:
I found out why you receiving a wrong measurement with a incorrect date but I am not sure why not receiving the correct measurement (but I have a an idea why it is not working)

Can you try the following:

1. Install the openScale beta version in the attachment
2. weight yourself
3. wait 5 min (don't do anything with your smartphone)
4. sync it with openScale (if you don't see a correct measurement try once again step 1 to 4)

[openScale-1.5.1_MiScaleFixScaleV2.apk.zip](https://github.com/oliexdev/openScale/files/1015149/openScale-1.5.1_MiScaleFixScaleV2.apk.zip)

kapitaniglo11, 2017-05-19:
Aaaand...

It works! So far! I'm uninstalling Mi Fit and I will try to use just openScale now.

Thanks for prompt fixing it! I hope I helped as well. Now to popularise your app... 😁

oliexdev, 2017-05-21:
great thanks for your help :+1: 

The problem was that the mi scale sends via Bluetooth advertising frame continuous your weight while your are weighting but with an incorrect date and time. Now I filter this advertising frames out.
Somehow after your weight (and I am not sure when exactly) the mi scale save the information in the advertising frame into the mi scale internal ROM with the correct date and time. 

Where the mi scale get the correct date and time is not clear for me because in the advertising frame has still the wrong date and time. After all there is still a missing link between how the mi scale store the weighting correct into the internal ROM!?

Do you get your measurement instantly after your weighting? Precisely after the mi scale blinks 3 times and shutdowns and then you sync with openScale? Or do you have to wait a longer period?

P.S.: There are many ways to increase the popularity of the app to help in developing, translations, to report bugs, enhancements, to spread it on forums, blogs or to rate the app on GooglePlay. Everything is welcome :smiley:

oliexdev, 2017-05-21:
I discovered with the new version of wireshark It is much more convenience to analyse the Bluetooth traffic. :smile: 

Now I found out how to get only the recently history weight measurements. I updated the [Xiaomi Mi Scale Wiki](https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale) respectively.

I hope the mi scale syncing process works much better now.

closed 2017-05-21 by oliexdev

kapitaniglo11, 2017-05-22:
So far it's working like a charm.

I have connected it to scale, _then_ stepped on the scale - immediate result in your app.

Thanks!

oliexdev, 2017-05-22:
with which version?
Did you try version openScale-1.5.1_MiScaleFixDateV3.apk?

kapitaniglo11, 2017-05-22:
Yes, with openScale-1.5.1_MiScaleFixDateV3

momo42, 2017-05-26:
The date bug seems to be fixed in openScale 1.5.2. Thanks for that. :-) But unfortunately we have another (probably related) bug now. I've opened another issue for that: https://github.com/oliexdev/openScale/issues/40