oliexdev/openScale #40: MiScale and two phones: syncing 'deletes' measurements

momo42, 2017-05-26:
Hello oliexdev,

currently I'm on openScale 1.5.2 (via F-Droid) and have the following problems.
My wife and me have two phones. On every phone we have two openScale users to use the smart assignment function of openScale.

In the morning we both stepped on our MiScale (me at 6:15 am, my wife at 6:45am). After breakfast (~ 7:30 am) we tried to sync our phones to get the new values. But unfortunately only my wife got the two new values. It seems like only the first phone that syncs gets all the measurements and after that they are gone. To check this behavior I stepped on the scale again and after that my own phone was able to receive this fresh measurement. But when I synced with my wifes phone after that, there were no new measurements again.

I also tried to remove the batteries of the Scale. Then I synced the Scale with mi wifes phone and got a lot of very old measurements (as expected). But when I synced with my own phone after that, there were no measurements again. I had no more time to check but I bet that if I reset the batteries again and then do the initial sync with my phone, my phone will get all the old values.

I checked both users on both phones. At this time both phones have a complete disjunct set of measurements.
This behavior is not very practical for using the scale with two persons - as most families do.

I think this could be an side effect of the fixed time bug. Would you please try to find a better multi user solution?

Thanks for your time!

Regards
momo

oliexdev, 2017-05-27:
Yeah it seems it is a side effect of the fixed time bug. Can you test the attached version of openScale?

[openScale-1.5.2-MultiUserFix.apk.zip](https://github.com/oliexdev/openScale/files/1033569/openScale-1.5.2-MultiUserFix.apk.zip)

momo42, 2017-06-02:
Thanks for the test version and sorry for the delay. I had a stress-filled week. My wife and me have now tested the MultiUserFix.apk and with it both phones we get all the values. So far, so good. :)

Here is how we tested:

- We uninstalled the regular versions and installed your openScale-1.5.2-MultiUserFix.apk.zip on both phones.
- Then we created two users on both phones and activated the smart assignment function.
- Now we took out and reset the scales batteries.
- The first sync (on both phones) didn't load ANY values.
- The second sync (on both phones) loaded ALL stored values to the selected user a.
- Then we manually deleted all wrong assigned values on both phones (very time-consuming by the way)
- The third sync (again on both phones) loaded the beforehand deleted values to user b.

After that both phones were constantly able to receive new measurements for both users. So yeah, it works! Please put your changes into the official version! Thanks for that!

I also detected some minor glitches in my tests and I have some ideas for improvements but I'll open new issues for that.

oliexdev, 2017-06-04:
> Thanks for the test version and sorry for the delay. I had a stress-filled week. My wife and me have now tested the MultiUserFix.apk and with it both phones get all the values. So far, so good. :)

awesome then I will close this issue.

> Then we created two users on both phones and activated the smart assignment function.

please see issue #43 how you properly do it.

> Now we took out and reset the scales batteries.

This is not necessary. You have to initialize the scale again to get the correct date.

> The first sync (on both phones) didn't load ANY values.

Because you took out the batteries :wink: openScale have to initialize the scale again. 

> The second sync (on both phones) loaded ALL stored values to the selected user a.
Then we manually deleted all wrong assigned values on both phones (very time-consuming by the way)
The third sync (again on both phones) loaded the beforehand deleted values to user b.

see again issue #43 to understand the problem.

> After that both phones were constantly able to receive new measurements for both users. So yeah, it works! Please put your changes into the official version! Thanks for that!

Great I commit the changes with commit https://github.com/oliexdev/openScale/commit/6c6737f42423d1d77961040d8b1dc4c8fbfe3b8c into the repository master tree

> I also detected some minor glitches in my tests and I have some ideas for improvements but I'll open new issues for that.

Nice so we can improve little by little openScale :smile:

closed 2017-06-04 by oliexdev