oliexdev/openScale #42: Cannot connect with Sanitas SBF-70 - No Bluetooth Device found

famo, 2017-05-30:
For some reason I cannot connect to my SBF-70.
Are there any special steps required?

System Info:
Sony Xperia Z3c, Android 6.0.1
Bluetooth Info:
[image]
[image]

oliexdev, 2017-06-02:
I don't own this scale. It was reverse engineered by @jflesch 
But your hardware address of the scale starts differently. Maybe this is the problem. 
Can you try the attached version of openScale?

[openScale_SBF70_NoHwCheck.apk.zip](https://github.com/oliexdev/openScale/files/1048260/openScale_SBF70_NoHwCheck.apk.zip)

jflesch, 2017-06-02:
The others UUID seems to fit, so assuming your screenshot is truncated, just adding your hardware address prefix should hopefully make it work.

famo, 2017-06-06:
Thanks for the reply.
Unfortunately the installation of openScale_SBF70_NoHwCheck.apk aborts - no reason given... :-(

nykroy, 2017-06-06:
I have the same scale and i cant connect that. 

Thank you for the app its a very good idea.

jflesch, 2017-06-06:
Hm, 3 questions:
1) Can you tell us exactly what popups you get ? ("Connecting ...", etc)
2) Did you set the scale name correctly ? If it's like mine, it's "SANITAS sbf70" (regarding the sbf-70, it is not case-sensitive)
3) While trying to connect to the scale, can you get the debug output with adb please : ```adb -d logcat BluetoothSanitasSbf70:D \*:S``` ?

nykroy, 2017-06-06:
Yes!!!! I add the scale name 'SANITAS sbf70' with the apk provided by @oliexdev and the app connect with the scale!!! Thank you very much.

famo, 2017-06-07:
Well, that the name has to match is an important information, which is not noticeable form the GUI at all!...
But yeah, when I change the name it works with the provided app.

However so far it only transfers the weight information, not water or muscle percentage.
How is it for you @nykroy ?

oliexdev, 2017-06-07:
> Well, that the name has to match is an important information, which is not noticeable form the GUI at all!..

Well what did you expect to enter by device name? But I already modified openScale so that on every device type a defaut device name is set.

closed 2017-06-07 by oliexdev

oliexdev, 2017-06-15:
@jflesch or @all
can somebody send me (olie.xdev@googlemail.com) or post here a picture of the Sanitas SBF70 scale for the [openScale wiki page](https://github.com/oliexdev/openScale/wiki) (same camera angle like the others)? 
I don't want to use an official image by the company due to copyright issues.

nykroy, 2017-06-15:
Yes, in 5 minutes i send you the photo

nykroy, 2017-06-15:
Check your email. If you need another photo tell me ok?