oliexdev/openScale #43: MiScale initial sync: Better way to assign values to users?

momo42, 2017-06-03:
While testing version 1.5.2 I stumbled upon an annoying problem:

If you use your scale with more than one person you have to set additional openScale user(s) to use the smart assignment for filtering away unwanted measurements.

But when you initially sync your MiScale, ALL stored values are synced to the currently active openScale user and you have to manually delete A LOT OF measurements after that. I tried it with two users but I don't know what do do if there are more family members using the scale.

Is there another (less time-consuming) way to assign the values to the users? Our scale is almost two months in use and I had to delete virtually hundreds of values. What happens if one installs openscale and there are thousands of values stored on the family scale? ... 
There must be a better way to achieve this separation... (multiple cut paste for example, or selecting and moving all values in a specified weight range).

The current behavior works but it can be extremely exhausting for new openScale users to sync their scale if their family used the MiScale for months/years before.

I think MiFit simply ignores all old values when you register a new user and asks you to step on the scale to 'connect' your user to the right weight range. In fact the users of MiFit even don't know that their scale provides a history of old values. I think most of them would be satisfied if openScale behaves like MiFit in this point - at least as the default.

oliexdev, 2017-06-04:
I must admit the smart user assignment in openScale is not documented and not very intuitive right now. :flushed: 
You don't have to manually delete any measurements.

Try this:

1. Create two or more users in openScale
2. Create for each user manually the first measurement with their respective weight
3. Synchronize your scale while the smart user assignment is on

Currently you have to manually add the first weight for each user otherwise openScale doesn't know the user initial weight and can't assign to the correct user.
I should add an "initial weight" field in the new user dialog. Then it would be more intuitive and you don't have to set the first weight manually.
But all measurements that can't assign to an user (e.g. the weight are out of range) are automatically stored to the current user. Of course I could ignore those measurements but then measurements will be hidden and the user don't have full control of their health data anymore but I will consider to create an "ignore out of range" option.

momo42, 2017-06-04:
Thank you for this explanation. I didn't even try to use the big blue button on the bottom yet. :grimacing: 
Giving openScale some initial values makes absolutely sense.

I think an 'initial weight' field in the new user dialog is a very good idea. I would also like to have an 'ignore out of range'-button. Perhaps you could put it direct below the 'smart assignment'-button and grey it out when 'smart assignment' isn't checked.

oliexdev, 2017-06-10:
I added with commit https://github.com/oliexdev/openScale/commit/870d6ed08ec05be436691d055efdd5996d59a925 a "ignore data out of range" option and with commit https://github.com/oliexdev/openScale/commit/13f45072fdc97800055ea9e69eb3f62373e863d7 I introduce a initial weight field to avoid problems when no first weight data exist for an user.

I hope this improve the usability of the app.

closed 2017-06-10 by oliexdev