oliexdev/openScale #46: Improvement: average weight per day/month (optional)

momo42, 2017-06-03:
While testing openScale me and my wife did a lot of measurements but also at normal usage it could happen that you step on the scale two or more times a day. But when you do so, the diagrams become very confusing. Here are two examples:
[image]
[image]
In my examples I already disabled the point notes to make the graph better visible. In my opinion more than one point per day is too much.

So please add options to use average values for one day (in month view diagram) and average values for one month (in year view diagram).

I would also like to have an option to set the diagram on the overview page to 'average per day' values. This would lead to one point per day. The default (one point per measurement) is absolutely ok and correct but if you often step on the scale more than once per day (for testing or for another reason) this diagram loses its usefulness. Example (6 of 7 measurements within two hours):
[image]

oliexdev, 2017-06-04:
Good idea to have one average measurement for a day but first I need to figure out how to implement that in a good way in openScale.

oliexdev, 2017-06-18:
implement an option to average data points for a day/month with commit https://github.com/oliexdev/openScale/commit/cadedb9a6009943a1891d72a53d5c28148d9efe9

closed 2017-06-18 by oliexdev

momo42, 2017-06-18:
Thanks a lot! I would really like to see a new 'official' version of the improved app in f-droid. The last release is 4 weeks old already. :smiley: 
Have a nice week!

oliexdev, 2017-06-19:
Yeah but I think the developer version has some bugs see #50  
Would be great if you could also test the developer version but create before a backup of your data.