oliexdev/openScale #50: Bluetooth scales debug testing

oliexdev, 2017-06-18:
I refactored the Bluetooth classes with commit https://github.com/oliexdev/openScale/commit/18a84570e4650df35c96cf14f7f2988b5f3e1ebe and https://github.com/oliexdev/openScale/commit/490061f7b8adbe1f5d09b5f438eaac1a1705b3cb so it will be more developer friendly to implement a new Bluetooth scale class. But I am not sure if I mixed up some source code... I hope not :sweat_smile:  
It would be great if somebody could test the [latest developer build](https://github.com/oliexdev/openScale/blob/master/openScale-dev-build.apk) and report any bugs or if everything works fine.

@jflesch I didn't refactored much at your source code but I am not sure if it still works because I set the BluetoothSanitasGattCallback in the constructor. Maybe somebody can check the Sanitas SBF70 scale.

Thanks in advance
oli

nykroy, 2017-06-18:
When i open the bluethoot settings the app crash

oliexdev, 2017-06-18:
@nykroy thx for the hint it should be fixed now please try again with the new developer build version.

nykroy, 2017-06-18:
Continue with the same problem :s

oliexdev, 2017-06-19:
I cannot reproduce this error any more :/
Did you uninstall and install the developer build version to delete all caches (make before a backup of your data)?

nykroy, 2017-06-19:
Yes i did. :s

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

nykroy, 2017-06-19:
May you add some code for catch the error when i executed the app and then i could send you the report, or if you said me how can i report you the crash i can help you. Maybe one try catch on the bluethoot link??

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

oliexdev, 2017-06-19:
Thx this week I am not at home so I can't debug it... 
Can you install android studio or use adb logcat to extract the stack trace with `adb logcat *:W`?

nykroy, 2017-06-19:
here is the log. Thank you for all your work!!!
[adblog.txt](https://github.com/oliexdev/openScale/files/1085735/adblog.txt)

oliexdev, 2017-06-19:
What makes me wonder is that the error points to line 126 in your adb log file but with commit https://github.com/oliexdev/openScale/commit/c601033e365714d4bfd93babc2c5f85164d84c59 I added one command so now it point to an empty line. I seems you didn't install the latest developer build or the last build didn't use my fixed source code. I just triggered travis to build a new developer APK file. The new apk file has the comment `openScale dev build 8 by Travis CI [skip ci]` please try this one.

nykroy, 2017-06-19:
Whith the new app no crash!!! Thank you

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

nykroy, 2017-06-20:
Bad news. Whit the last apk i select the scale sanitas sbf70, theni go back to the home, i  push the bt button but dont find my scale. If i go again to the bluetooth setting i can see that the bt setting havent been saved O_o. If you go to bt setting you allways have selected the custom Open scale.

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jflesch, 2017-06-20:
FYI, bt scale data history is not supported with the SBF-70

nykroy, 2017-06-20:
I know it but i cant select that xD

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jflesch, 2017-06-20:
My message was actually for @oliexdev :)

nykroy, 2017-06-20:
Ok jejeje sorry i didnt understood you 

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jflesch, 2017-06-21:
OK, I've tested and connection to the SBF-70 works for me.
However, I'm only getting the weight, not the other values.

jflesch, 2017-06-21:
Nevermind. It's still working for me. (for some reason, the first time I tried, the scale returned 0 on all the values but the weight ....)

jflesch, 2017-06-21:
@oliexdev OK, changes look good to me :)

jflesch, 2017-06-21:
@oliexdev However, I have the same problem than @nykroy . It appears BT settings are correctly saved, but are not displayed back correctly.

nykroy, 2017-06-21:
I dont now what the problem it is but the scale is on. I add a screenshots. :s
[image]

jflesch, 2017-06-21:
There are no warnings in the logs you posted. Can you try with something a bit different please :
```sh
adb -d logcat BluetoothSanitasSbf70:D \*:S
```

jflesch, 2017-06-21:
Or even better:

```sh
./adb logcat BluetoothSanitasSbf70:D BluetoothMiScale:D BluetoothCommunication:D \*:S
```

nykroy, 2017-06-21:
of course. Thank you very much!!!
[No bluetooth device found.txt](https://github.com/oliexdev/openScale/files/1092616/No.bluetooth.device.found.txt)

jflesch, 2017-06-21:
Actually, while reading your logs, I realized that the address of your scale simply has a prefix unknown to OpenScale yet. I assume OpenScale has never worked with your scale yet ?

Anyway, I've added it to my PR for @oliexdev : https://github.com/oliexdev/openScale/pull/51

nykroy, 2017-06-21:
With the openScale_SBF70_NoHwCheck.apk i havent any problem to connect

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jflesch, 2017-06-21:
(I assume you mean the .apk built for #42 ?)
Well .. to be perfectly honest, I think it shouldn't have worked since the prefix of the BT address doesn't match. However, I'm not sure what was the exact modification included by @oliexdev . Maybe he disabled the whole address check entirely for this test apk ?

nykroy, 2017-06-21:
Yes with that apk it works for me and my wife phone :s

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

oliexdev, 2017-06-21:
Yes I disabled the whole address check in the openScale_SBF70_NoHwCheck.apk test build.
Thx @jflesch for the PR #51 
The device in BT setting is always reset because of https://github.com/oliexdev/openScale/commit/c601033e365714d4bfd93babc2c5f85164d84c59 
In this commit I set the default value to zero which is custom openScale device. I set it because otherwise it crashed on a clean start-up. I need to fix it but currently I am on a business trip and on the weekend I will be busy so it could take some time.

nykroy, 2017-06-21:
Dont worry @oliexdev you are doing a very hard work for the sbf70 users because the original app doesnt work fine. I am very grateful with you and @jflesch 

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jflesch, 2017-06-21:
@oliexdev Just a thought : There seems to be quite a lot of address prefixes for the Sanitas SBF-70. I wouldn't be surprised if we discover more later. For this scale, wouldn't it be better to simply not filter on the BT address at all but only on the BT name ?

nykroy, 2017-06-21:
I install the last apk was modified by @oliexdev some minutes ago and now the scale is working!!

_Sent from my LGE LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

oliexdev, 2017-06-30:
I fixed the Bluetooth preference bug with commit https://github.com/oliexdev/openScale/commit/c97ae5376d3b0803ea91cd3d714eaba2e244fdf9

@jflesch I tried it once to disable the hw address check at all but openScale crashed. I found out now that if no device is found an empty string is returned. Now I will check this string so for SBF70 scale the hw address check is disabled. 

Ok great I think the refactoring is working thx @all.

closed 2017-06-30 by oliexdev