oliexdev/openScale #52: can not connect to sanitas sbf70

torpak, 2017-06-24:
I set up the scale using the manufacturers app and then tried to connect with open scale using the sanitas sbf70 profile from the bluetooth menu i even set the scale name to "SANITAS SBF70".
No matter if the scale is active or inactive when i start it allways says it can't find the scale.
(I stopped the manufacturer app before to make sure it doesn't interfere)

oliexdev, 2017-06-30:
Hi torpak,

I disabled the hw address check for the SBF70 please try the latest developer apk build 14

Best regards
oli

torpak, 2017-07-01:
That is mich better. The connection works mostly now.
Sometimes the weight is detected as zero. But i can live with that. Thank you!

closed 2017-07-02 by oliexdev

roqeer, 2017-10-04:
I have similar problem. New scale SBF70 worked great with OpenScale but returned only weight without other measurements. hone with Android 7.1
Then I set up with original app to add user profiles with other device on Android 6.0 (original app is rather bad and doesn't work on 7.1). This allows measuring additional things.
After that OpenScale stopped working with SBF70. It shows that it connects to scale but the scale doesn't show bluetooth connection icon and no measurements can be made in OpenScale. On the SBF70 all measurements are saved.
Any idea how to set up so that I can log all measurements in OpenScale?
Original app is really bad and nothing compared to OpenScale.

oliexdev, 2017-10-06:
I don't own this scale so I can't really help but if I interpret the source code correct the user name in openScale have to be same as your added profile in the original app. Other idea is to reset your scale? Delete all profiles and let openScale add one. But make before a backup of your data. Just some ideas.
Please give feedback if one works.

timoc, 2017-11-11:
I have the same issue - the FAQ says "Please step onto the scale barefoot and ensure that a user has been added and assigned. If user assignment has taken place, your initials will be shown on the scale display. If this is not the case, perform user assignment in the app in the menu area Settings > My devices > SBF 70.  "
Does openScale create users?

oliexdev, 2017-11-11:
@timoc I didn't implemented it but as I can see from the source code yes an user is created but only if no user exist on the scale and if the maximum users are not reached. 

So my suggestion is to delete all users using the original app and then use openScale (make a backup of your data first!)

Please give feedback if this is working.

lenucksi, 2017-12-25:
Sorry to reply to an open existing, closed issue, however my issue pretty much follows up on this one. Feel free to decouple it to it's own, new issue.
I've got the same scale and have the following behaviour:
 - Scale reset using the button on the bottom side
 - Smartphone using 7.1
 - Last revision of OpenScale out of the F-Droid Store
 - The vendor app recognizes the scale, however fails to properly add and configure it, remaining in a loop "downloading" at the "add device" stage.
 - OpenScale immediately finds and connects to the scale and properly measures the weight and only the weight.
 - Once OpenScale is installed, the vendor app is not even able to find and connect to the scale, however the scale shows toasts with the current weight while attempting to configure. It is required to step onto the scale to add it to either app.
 - Using OpenScale some pseudo-user appears to be created, however even though a name is configured, the scale only shows '---' as username.

On an older version of Android using the vendors app it was successfully possible to fetch all measurements into the app. However, as @roqeer said, the app is really, really bad. It even failed to connect to the scale once the scale history memory got full.

I'd love to be able to fetch the other metrics or even use the history function of the scale. Can I assist in any way to allow this to be debugged?