oliexdev/openScale #54: BS444 no bluetooth device found

kuehnelth, 2017-07-11:
Hi,
I just installed the app from Google Play (v1.5.3) but I cannot connect to my scale.
Probably because the MAC-address is not in the whitelist. It starts with C9:A6:8A.
Could you add that address?

Thanks

oliexdev, 2017-07-12:
thx for the hint. I added the new hw address. Please install the latest development build 21.

closed 2017-07-12 by oliexdev

vogan, 2017-07-21:
Hi,

I had the same issue, different MAC-address. Fixed it by adding:

`hwAddresses.add("D60211");`

Thank you!

oliexdev, 2017-07-23:
Hi vogan,

thanks for adding it. If you want you can create a pull request, so you will listed as a contributor. If not no problem then I will add it.

Best regards
oli

slobinger, 2017-08-02:
Same issue here. Added my hdAddress and made pull request #64.

Tanks for your great work.
Best regards
Sebastian

hirnsushi, 2017-08-09:
Same problem here.
My MAC-Address starts with DA:45:CF.
I would have opened a pull request myself but wanted to make sure I really only have to include the beginning of the address.

hirnsushi, 2017-08-09:
That would be great, thx.

oliexdev, 2017-08-09:
it's done with commit https://github.com/oliexdev/openScale/commit/94104497beb009c71cc03e80ef97e5157b2327b8
Please install the latest developer build 41 to get this change. The hw address will official contain in the app after the next release.