oliexdev/openScale #58: Initial preset of measurement settings not regarded

MilenaMMay, 2017-07-25:
When I load the app to a new (virtual) device and create a new user,
the Overview view contains all measurement settings.

When I then go to Settings -> Measurements,
I see that waist and hip circumferences are disabled by default.

When I then close the Settings without changing anything and go back to Overview, 
only the measurements that are enabled are shown (waist and hip circumferences are not shown).

This can only be reproduced by loading the app to a new (virtual) device and not by creating a new user because the settings are global.

oliexdev, 2017-07-29:
thx it is fixed some default values were on true instead of false.

closed 2017-07-29 by oliexdev

MilenaMMay, 2017-07-31:
Great, thx =)