oliexdev/openScale #60: Bounty for mi scale 2 support

kxpvcr, 2017-07-25:
Hey! Just tested this with the 1st gen. scale and it works just fine. Do you have any idea of how much effort to support the 2nd gen. scale?

I can set up a bounty and contribute to it. Maybe people are into that, it'd be nice to have those extra measurements.

What do you think?

oliexdev, 2017-07-29:
Hi kxpvcr,

I have no idea how much effort is to reverse engineer the 2nd generation of the mi scale. Maybe they change not much but it could also be possible if they encrypt all Bluetooth messages.

Thanks for your offer maybe somebody will accept it. For me I don't want any bounty (if you like openScale give it a positive rating). Furthermore here in Germany you can't buy the 2nd generation of the mi scale only as an expensive import from China. 
But I will do my holidays in South Korea. Maybe I will buy one for testing purpose ;)

Best regards
oli

closed 2017-08-09 by oliexdev

kxpvcr, 2017-08-25:
Stellar 🥇 

I got mine from Spain, not that expensive. If you decide you'd like some monetary compensation for your effort, let me know!

nykroy, 2017-08-25:
Me too, i searched time ago to donate some money for your efforts but i didnt find anything

_Sent from my Lge LG-H870 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

NerosTie, 2017-10-05:
Hey! Any news about the support of the Mi Scale 2? Thanks

oliexdev, 2017-10-06:
Yeah yesterday I got an email that the package was sent to me but it comes from China so it takes some time.

NerosTie, 2017-10-06:
Nice! Because my Mi Scale 1 just died after 2 years so maybe I'll take the new version...

mNantern, 2018-04-21:
Hey ! And thank you for that nice app' !

Any news about the full support of Mi Scale 2 ? How can we help ?

erijo, 2018-04-21:
The best way to help would be if someone could reverse engineer the original app to learn how the calculations are done.