oliexdev/openScale #63: Copyright violation

Poussinou, 2017-07-31:
Hi @oliexdev 

I found [this app on the Google Play Store](https://play.google.com/store/apps/details?id=com.weight.watcherLoss). This guy just took the source code, added some adds and put it on the Google Play Store

As you are the owner of the source code, feel free to open a request to google [here](https://support.google.com/legal/contact/lr_dmca?product=googleplay) if it's your wish ;)

Don't forget to tell Google that:

* The app is licensed under GPL3 and this guy is violating it by distributing the app without the source code;
* He is falsely claiming he made the app;
* Also tell them that this guy has done the same thing already with other apps. Probably this is worth mentioning. (I watch him)

oliexdev, 2017-08-01:
thx it has been reported to Google. We will see how fast Google is to remove this app. He published it also on Amazon...

Poussinou, 2017-08-01:
Do not hesitate to send a mail to Amazon^^
Yep we will see how fast they are...

oliexdev, 2017-08-09:
Now Google is working to remove the WeightWatcher app by [TopMobile Tech](https://play.google.com/store/apps/dev?id=8641344094929372114) from GooglePlay.

I also stated now a DMCA takedown notice to Amazon regarding to the [WeightWatcher app on Amazon](https://www.amazon.com/TopMobile-Tech-Weight-Watchers-BMI/dp/B073ZPRH9J)

oliexdev, 2017-08-10:
Amazon was pretty quick. On Amazon and GooglePlay the WeightWatcher app is now removed.

thanks again @Poussinou you are really a top GPL violation hunter :1st_place_medal:  

I see you reported many developer about open source violations. The open source community need more people like you. Do you have a good strategy to find those apps that are violating the GPL?

closed 2017-08-10 by oliexdev

Poussinou, 2017-08-10:
Thanks for taking some time to report them :smiley: 
No problem !
I have a good search strategy yep, but I do not explain it in public since it could help fake developers to circumvent it... But I have no script nor software to do it, nothing can replace a human judgment in this case, I assume.
My only tool is a document where I keep a log of fake developers, apps I reported, stuffs like that...
Do not hesitate to hunt this kind of violation by yourself for your own apps. Then report them here, so that I can track the "developers". If you want me to check if there is any violation on the Play Store for a specific app, just ping^^

Poussinou, 2017-08-10:
I do not "hunt" on the Amazon Store yet, but I'll maybe do now

oliexdev, 2017-08-26:
you are right your strategy must stay secret :smile: 
But without scripts or software must be really hard to find those violations. 
An image matching algorithm would be good to find at least first hints of a violations just by comparing  the original app screenshot with all other app screenshots on Google Play... :thinking:

Poussinou, 2017-08-26:
Yep but I'm not good enough to code such a script... ^^
I was on holidays, I'll hunt a little bit now