oliexdev/openScale #65: BLE Scanner for selecting a BLE device

oliexdev, 2017-08-02:
openScale currently requires a hardware address or a device name for the Bluetooth connection.
Unfortunately, the Medisana BS444 scale always has different hardware addresses and device names, see #54.

How is the best and user friendly way to handle with different hardware addresses and device names of scales? Maybe somebody is a Bluetooth expert or as an idea to solve this issue.

kuehnelth, 2017-08-17:
The official app for the BS444 seems to search for the service UUID "000078b2-0000-1000-8000-00805f9b34fb" to decide if a BLE device is a BS444 scale or something else.
So maybe you could also use this method to identify it?

oliexdev, 2017-08-26:
Would be possible but then I have to discover all Bluetooth devices and UUIDs.

oliexdev, 2017-10-01:
implemented a BLE scanner to select a device, see commit https://github.com/oliexdev/openScale/commit/ba9f72d5ee3a9c098a7df421d647b259dd53f3e7.

Maybe somebody can test that with their scale if it works?

closed 2017-10-01 by oliexdev