oliexdev/openScale #67: New Launcher Icon?

DavidXochipilli, 2017-08-26:
Hey @oliexdev,
thanks you so much for your awesome app! Are you interested in a new launcher icon?
I am not good at design myself but I could ask someone. Do you have any wishes?

oliexdev, 2017-08-26:
Hi DavidXochipilli,

I am very interested in a new launcher icon. :smiley: I am not a good graphic designer as well. 
I have only two wishes: Firstly the icon launcher should be available in the following sizes to met the android design requirements:
512x512px,  192x192px, 144x144px, 96x96px, 72x72px, 48x48px or in a svg format.
Secondly the icon must be licensed in a way that I can use it in my app, see for example https://creativecommons.org/ for some licenses (the authors name will be of course included in the CREDITS)
Just for your information Google has published a design guide for icons https://material.io/guidelines/style/icons.html#icons-product-icons maybe it helps a bit.
My used icon colour is `#33b6ea` which should be somehow used in the new icon. 

Best regards from Germany
oli

DavidXochipilli, 2017-08-27:
@Sajidshaik will do the design and I will send him a few dollars for his work 👌

DavidXochipilli, 2017-08-28:
Here's a preview @oliexdev, please tell us your thoughts:
[image]

oliexdev, 2017-08-28:
Good job it looks already better than the current openScale icon :+1: 

I think the design of the new icon is a bit too "complex". For example in a small icon scale you can't really see the black dots anymore. 

Some app icons maybe for inspiration which I like are:
https://play.google.com/store/apps/details?id=jp.co.recruit.mtl.beslim
https://play.google.com/store/apps/details?id=com.yunmai.scaleen
https://play.google.com/store/apps/details?id=com.ikdong.weight
https://play.google.com/store/apps/details?id=com.dencreak.weightwar
https://play.google.com/store/apps/details?id=org.fmouri.weightLog
https://play.google.com/store/apps/details?id=com.medm.app.health
https://play.google.com/store/apps/details?id=com.despdev.weight_loss_calculator

But it is definitely only my personal opinion. I think often in design less is more. But I really like the shadow play in the new icon.

DavidXochipilli, 2017-08-29:
I will send two icon versions, one with the dots and one without.
Is that fine or do you have other ideas? 🤔

oliexdev, 2017-08-30:
I don't have really a good idea. Maybe the icon designer has another idea or somebody else in the openScale community?

But anyway I will use also the icon version above if nobody has another idea. But there is also no time pressure because I will be in the next weeks anyway on my holidays and travel around the world :smile:

DavidXochipilli, 2017-08-30:
@Sajidshaik and I don't have another good idea either. 😥
He will upload a .zip with your requested sizes and the .ai source file in some days.
Maybe the community can give us some input, they are our last hope. 😀
We wish you an awesome journey! 😎

oliexdev, 2017-08-31:
Thank you!

I created a icon based on https://www.flaticon.com/free-icon/weight_527064 by Those Icons

What do you think?

[image]

DavidXochipilli, 2017-09-04:
To be honest, I don't like your icon that much because it isn't really Material. 🤔

nykroy, 2017-09-04:
I think the same that @DavidXochipilli

DavidXochipilli, 2017-09-16:
I attached a zip with the icon.
For the [Credits](https://github.com/oliexdev/openScale/blob/master/CREDITS): [Sajid Shaik](https://plus.google.com/u/0/+DroidAppsReviewer).
[OpenScale.zip](https://github.com/oliexdev/openScale/files/1308578/OpenScale.zip)

Sajidshaik, 2017-09-17:
Here you go @oliexdev 
[OpenScale.zip](https://github.com/oliexdev/openScale/files/1308882/OpenScale.zip)

oliexdev, 2017-09-21:
I am back from my journey, I had a great time :sunglasses: 

Thank you so much for your designed icon @Sajidshaik and @DavidXochipilli for your effort but honestly currently I am a bit uncertain which icon I should use... :thinking: 

What do you mean it isn't really material @DavidXochipilli?

oliexdev, 2017-09-22:
Btw @Sajidshaik under which license is your icon?

DavidXochipilli, 2017-09-23:
> I am back from my journey, I had a great time 😎

That are great news! 👍

> Product icon design is inspired by the tactile and physical quality of material. Each icon is cut, folded, and lit as paper would be, but represented by simple graphic elements. The quality of the material is sturdy, with clean folds and crisp edges. The matte-like finish interacts with light through subtle highlights and consistent shadows.

[https://material.io/guidelines/](https://material.io/guidelines/)
[https://material.io/guidelines/style/icons.html](https://material.io/guidelines/style/icons.html)
[Making Material Design on YouTube](https://youtu.be/rrT6v5sOwJg)

Maybe @Sajidshaik can add a better explanation. It is quite difficult to explain for me. 🤔
Maybe even [Google](https://www.youtube.com/channel/UClKO7be7O9cUGL94PHnAeOA) does not know it exactly. 🤤

oliexdev, 2017-09-23:
I understand what you mean. I've now decided on an icon, I think the below icon fits more to openScale because of its simplicity but that doesn't mean Sajidshaik icon is bad. Hope your are not mad at me... :(

I have changed the icon a little bit but to comply with the material design guide, I think I am too bad at inkscape. 

[image]

svg-file

[launcher_openscale.svg.zip](https://github.com/oliexdev/openScale/files/1326659/launcher_openscale.svg.zip)

closed 2017-09-23 by oliexdev

DavidXochipilli, 2017-10-11:
The icon is under the CC0 1.0 Universal license.