oliexdev/openScale #68: Diagram year looking strange

john-bill, 2017-09-09:
Hi, after upgrading to 1.5.4 from f-droid the diagramm per year is looking strange, all values per month seem to be printed on first of the month.

[image]

PabloGN, 2017-09-14:
Yes, it is because the last change in android_app/app/src/main/java/com/health/openscale/gui/fragments/GraphFragment.java 
         if (field == Calendar.DAY_OF_MONTH) {
             day_date = new SimpleDateFormat("dd", Locale.getDefault());
        } else if (field == Calendar.MONTH) {
            day_date = new SimpleDateFormat("MMM", Locale.getDefault());
         }
And the x-axis:
-            generateLineData(Calendar.DAY_OF_YEAR);
+            generateLineData(Calendar.MONTH);
I don't see clearly whether it is a simple change on the axis, maybe when generating the points.

oliexdev, 2017-09-21:
In the settings->diagram is an option to enable or disable to calculate the average of a month / day, regarding to issue #46.
Did you enabled it? But even if it is enabled, the diagram looks a bit strange. Can you post the exported csv file so I can look into it?

john-bill, 2017-09-25:
When average setting is enabled, the diagram looks ok. But when disabling this setting I get the effect, that all monthly values are printed on the last day of month. 
csv as txt attached

[openScale_data.txt](https://github.com/oliexdev/openScale/files/1330488/openScale_data.txt)

oliexdev, 2017-09-26:
ok not really a bug. It was only that all data (if average data was disabled) was showed on the corresponding month. That is why you have such vertical lines but I improved it to show the days of the year with commit https://github.com/oliexdev/openScale/commit/bfea2d6c3e65b34123aa5a153c8c6c7f9f411a6c

closed 2017-09-26 by oliexdev