oliexdev/openScale #69: Linear regression for measurements

beothorn, 2017-09-21:
When trying to achieve a goal, it is nice to see to which direction the measurements are leading to, specially the weight.
I've used other weight measurements software that showed me how when I would reach my goal if I kept my current pace and it was really usefull.
Maybe as an option?
Is that something you would find usefull?

oliexdev, 2017-09-21:
Hi beothorn,

I think as an option it would be useful to have a linear regression line for the weight but for all measurements I think that would be too much...   Additionally I can add an optional "goal" line. 

Best regards
oli

oliexdev, 2017-09-22:
I implemented a goal line and a linear regression line for weight values with commit https://github.com/oliexdev/openScale/commit/8761ef5f0e0f74183567d47c1066df6cff0a4df0

But I am not really sure anymore if it really helps the linear regression line because depending on the data it is only rough outline of your predicted weight process!?
I tried also a quadratic regression line but is that more useful? Anyway I implemented as an option so anybody can enable or disable the two lines in the graph.

closed 2017-09-22 by oliexdev

oliexdev, 2017-09-22:
I added an option to set the polynomical order for the regression weight line.

beothorn, 2017-09-23:
Wow, thanks, that was fast :)

I'm using it to lose weight and I feel more motivated when I see my weight loss trend compared to my goal. 
It is not the best model to predict weight loss but because my goal is set to next year, it helps me to know if I am losing too much weight (I don't think it is healthy to lose too much in a too short time) or too little.

Quadratic is really nice, but at least for me I will keep Regression order at 1.

I saw that the goal line wasn't rendering because:
```
valuesGoalLine.push(new PointValue(0, goalWeight));
valuesGoalLine.push(new PointValue(31, goalWeight));
```
I've added some code at https://github.com/beothorn/openScale/commit/089a871375000e39ef9cfd014fffd1585ad20fe6
just so the goal line shows up.

It seems that the goal line and the prediction line aren't yet correct for the monthly view. I'll take a look.

Thanks for the awesome app

beothorn, 2017-09-23:
Ops, prediction line is correct (I've added entries only for one month) , just my code for the goal line it's not.

beothorn, 2017-09-23:
https://github.com/oliexdev/openScale/pull/70

oliexdev, 2017-09-23:
The goal line was displayed but I implemented it in a way that is not automatically zoomed out to the goal line because it is possible that the goal weight and the weight values have a big gap. In these situation the x-axis would be to big. So please zoom out the diagram to see the goal line.

I think we have different views of how a goal line should be displayed in the diagram.  :wink: 

For me its a horizontal straight line on your target weight, that you have "to pass" either from below or from above depending if you want to lose or gain weight.

Now you implemented a goal line which start at the last measurement to your goal date with the goal weight. The line is no longer straight anymore. For me it makes less sense.

beothorn, 2017-09-23:
Oh, I see. Yes, I misunderstood what you've meant with goal line  :)
I did what I did because then I can compare it to the prediction line, like this

[https://imgur.com/a/gAFnt](goal vs prediciton)

So, I think both makes sense, but in two different context. The straight line for when you are trying to maintain weight and the line towards a date when you are trying to lose weight.

Maybe both as different options?

oliexdev, 2017-09-23:
You are right both variants makes in some way sense but I think your interpretation of a goal line is more confusing. I will keep my implementation, sorry... :(
But thanks for your idea and your effort, that keeps me motivated.

beothorn, 2017-09-23:
No problem.
Thanks for your time :+1: