oliexdev/openScale #71: Reverse enginnering Yunmai mini

oliexdev, 2017-09-23:
I am trying to reverse engineer the yunmai mini scale.

I already found out the weight values but I can't figure out the body fat percentage, muscle percentage and so on. Any help or idea is welcome.

So far what I got:

- No weight values are stored on the scale itself. You have to be connected to the yunmai app to receive any data
- Weights are continuous send to UUID: Unknown (0xffe4) (if you stay on the scale)
- Sent values are like `0d:1f:0b:01:59:c6:15:2f:26:95:03`
  * byte 0-6: bytes are constant
  * byte 7: first nibble is constant, second nibble is changing
  * byte 8-9: contains the weight value (here `0x2695` = 9877 = 98,77 btw the value is rounded in the yunmai app as 98,8 kg)
  * byte 10: is changing
- last measurement from the scale of the continuous sent send data looks like `0d:1f:14:02:00:59:c6:15:2f:23:d0:b6:32:26:95:00:a5:08:f2:37`
  * byte 0-5: bytes are constant
  * byte 6: first nibble is constant, second nibble is changing
  * byte 7-8: bytes are changing
  * byte 9-12: bytes are constant
  * byte 13-14: contains the weight value again
  * byte 15: bytes are constant
  * byte 16-19: bytes are changing
- after that the yunmai app send some values to the scale. In response they send some values back to the scale like the following `0d:1f:12:06:01:23:d0:b6:32:af:01:19:55:5a:8d:26:01:6f`
-  I guess that the values displayed in the yunmai app are rounded values like as the weight value
- Since so many bytes are constant and only few bytes are changing it could be possible that the information of the body measurements (like fat, muscle, water percentage etc.) are compressed or somehow encrypted. (for example a fat percentage of 22,9 % is somehow in the above values encoded)

Here are my Bluetooth log files with the corresponding body measurements which we have to find (open it with wireshark):

**23.9.2017 07:36**
[btsnoop_hci_979.log](https://github.com/oliexdev/openScale/files/1326417/btsnoop_hci_979.log)
* Weight 97,9 kg
* Body fat 22,9 %
* Muscle 51,7 % 
* Water 56,0 % 
* Protein 16,2 %
* Bone mass 4,0 %

**23.9.2017 10:02** 
[btsnoop_hci_988.log](https://github.com/oliexdev/openScale/files/1326418/btsnoop_hci_988.log)
* Weight 98,8 kg
* Body fat 22,9%
* Muscle 51,7% 
* Water 56,0% 
* Protein 16,2%
* Bone mass 3,9%

**22.9.2017 18:35**
[btsnoop_hci_1041.log](https://github.com/oliexdev/openScale/files/1326419/btsnoop_hci_1041.log)
* Weight 104,1 kg
* Body fat 25,0 %
* Muscle 50,3% 
* Water 54,5% 
* Protein 16,1%
* Bone mass 3,8%

Compared last continous messages of all three values
```
 0d:1f:14:02:00:59:c5:f7:d6:23:d0:b6:32:26:3b:00:dd:08:f2:f9  // 97,9 kg; water 56,0 %
 0d:1f:14:02:00:59:c6:15:2f:23:d0:b6:32:26:95:00:a5:08:f2:37  // 98,8 kg; water 56,0 %
 0d:1f:14:02:00:59:c5:3b:d8:23:d0:b6:32:28:ae:00:c2:09:c4:88  // 104,1 kg; water 54,5 %
```

oliexdev, 2017-09-24:
I discovered that the last byte of each row is a XOR checksum starting from byte 1.

oliexdev, 2017-09-24:
ok byte 17-18 contains the body fat in percent multiplied by 100.

oliexdev, 2017-09-29:
thanks to @pwnall and @conoro already analysed the protocol under, see [here](https://gist.github.com/conoro/f0c1d96c450a8f5cce70e2846c3686c4)
Also thanks to Dodo who found that information.

So in my example we have now the following information:
```
0d:1f:14:02:00:59:c5:f7:d6:23:d0:b6:32:26:3b:00:dd:08:f2:f9  

    Weight 97,9 kg
    Body fat 22,9 %
    Muscle 51,7 %
    Water 56,0 %
    Protein 16,2 %
    Bone mass 4,0 %

02: finished weighing
00: historical info
59:c5:f7:d6 = unix time in seconds = 1506146261 = Saturday, 23 September 2017 05:57:41
23:d0:b6:32 = user ID 
26:3b = 9787 = 97.9 kg
00:dd = resistance which is apparently used to calculate muscle mass, body water etc.
08:f2 = 2290 = 22,9 % bodyfat
f9: checksum
```
But sadly I have no idea how to calculate the muscle, water percentage, bone mass and protein out of the resistance. I found some formula but none are close to the above values. :/

oliexdev, 2017-10-01:
implemented a first version of the Yunmai Mini (only supported weight and fat) with commit https://github.com/oliexdev/openScale/commit/c37f71939a613549c0532351a0605156b7b43ddf

Vanuan, 2017-10-31:
I have Yunmai SE shown as "YUNMAI-ISSE-US"
MAC: 60:64:05:xx:xx:xx (Vendor: Texas Instruments)

The protocol looks very similar. How can I force open scale to use Yunmai mini driver?

Vanuan, 2017-10-31:
```
services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       59554e4d41492d495353452d5553 | 'YUNMAI-ISSE-US'
  2a01 (Appearance)
    properties  read
    value       0000 | ''
  2a02 (Peripheral Privacy Flag)
    properties  read
    value       00 | ''
  2a03 (Reconnection Address)
    properties  write
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       5000a0000000e803 | 'P h'
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
ffe0
  ffe4 (Key Press State)
    properties  notify
ffe5
  ffe9 (Key Press State)
    properties  write
```

oliexdev, 2017-10-31:
If you want to try your scale with the Yunmai Mini protocol change the method `checkDeviceName(...)` in https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothYunmaiMini.java
to the following:
```
    @Override
    public boolean checkDeviceName(String btDeviceName) {
        if (btDeviceName.startsWith("YUNMAI-SIGNAL-M") || btDeviceName.startsWith("YUNMAI-ISSE")) {
            return true;
        }

        return false;
}
```

This will accepts your device name as a Yunmai Mini device.

Vanuan, 2017-10-31:
It looks like it is identical.
The only difference is that these 2 parameters are always `00 00 00 00`:

```
    .uint16be('resistance')
    .uint16be('fatPercentage')
```

oliexdev, 2017-10-31:
if I see it right on http://www.iyunmai.com/us/m/light-se/ the Yunmai SE scale can only measure your weight.
@Vanuan did you received your weight with the yunmai mini protocol? If yes I can implement it in openScale.

Vanuan, 2017-10-31:
Yes.

oliexdev, 2017-10-31:
I implemented a version of Yunmai SE. Please install and try the beta version of the openScale under https://github.com/oliexdev/openScale/releases

Vanuan, 2017-10-31:
Sure. Does this look good?
[image]

Vanuan, 2017-10-31:
It works. I see "[weight] to [user] added".
Can this app send some command so that units are changed from `lb` to `kg`?

oliexdev, 2017-11-01:
sure openScale can send commands but I need to know what I have to send to change the unit :wink: 
You have to analyse the Bluetooth protocol with wireshark see openScale wiki for more information how to do that.

Thanks for your image. Is it possible to take another one while the scale is turned off?

Vanuan, 2017-11-02:
[image]

zamicol, 2017-11-11:
After reading this conversation, I'm buying a Yunmai SE.

cantruchd, 2018-01-11:
Maybe we can use the body fat % to calculate the other factor?

MarvinT, 2018-01-20:
@oliexdev 

If we get a bunch of data points i.e. different peoples weights and resistances... or even weights and resistances for different times throughout the day we can figure out the equations. Probably doable with 6, fairly easy with 10, and very easy with 20 data points

I'm trying to decide which scale to buy now and probably going to use your app as the determining factor.

oliexdev, 2018-01-20:
@MarvinT 

Yeah the idea is not new. You can try it but be aware that the formula will probably change if you are  male or female and it will be probably also depends on your body height, age and the impedance value.

So you have to consider a few variables. 

Maybe the issue #154 helps you for your decision.

dylanlt, 2018-04-09:
Total Body Water (TBW) = (0.372 * HeightCM * HeightCM / Resistance)  + (0.142 * massKg) - (0.069 * ageYears) + (3.05 * ifMale)

Fat Free Mass = TBW ÷ 0.73 

Can you test with your height and age and see how well it compares to the app value?

Reference: LUKASKI and BOLONCHUK (1988) Formula for total body water. Aviation Space and Environmental Medicine, 59, p. 1163-1169

oliexdev, 2018-04-10:
@dylanlt I already tried that formula but it didn't work out.
Try for yourself for the example measurements: 

used user data on all measurements:
`Height 5'9“ ft (175cm) Age 25 male`

example measurements 1:
```
0d:1f:14:02:00:59:c4:ff:cf:23:d0:b6:32:26:aa:00:9f:08:de:16  // 99,0 kg
22.9.2017 14:19  99,0 kg
Body fat 22,7
Muscle 51,8
Water 56,1
Protein 16,2
Bone mass 3,9
Resistance 0x9f (maybe!?)
```

example measurements 2:
```
0d:1f:14:02:00:59:c5:3b:d8:23:d0:b6:32:28:ae:00:c2:09:c4:88  // 104,1 kg
22.9.2017 18:35 104,1 kg 
Body fat 25,0
Muscle 50,3% 
Water 54,5% 
Protein 16,1
bone mass 3,8
Resistance 0xc2 (maybe!?)

```

example measurements 3:
```
0d:1f:14:02:00:59:c5:f7:d6:23:d0:b6:32:26:3b:00:dd:08:f2:f9  // 97,9 kg
23.9.2017 07:36 97,9 kg 
Body fat 22,9
Muscle 51,7% 
Water 56,0% 
Protein 16,2
bone mass 4,0
Resistance 0xdd (maybe!?)
```

example measurements 4:
```
0d:1f:14:02:00:59:c6:15:2f:23:d0:b6:32:26:95:00:a5:08:f2:37  // 98,8 kg
23.9.2017 10:02 98,8 kg 
Body fat 22,9
Muscle 51,7% 
Water 56,0% 
Protein 16,2
bone mass 3,9
Resistance 0xa5 (maybe!?)
```

dylanlt, 2018-04-13:
Typical population average resistance index (height squared divided by resistance) is roughly 60 a seen here: https://pdfs.semanticscholar.org/98a3/16bcdb1ce64790ca796278ed7731b204764c.pdf

Their formula has slightly different weightings, too. 

Working backwards though, fat 22.9% means that fat = 22.4kg, so FFM = 97.9 - 22.4kg = 75.5kg

Hydration constant is 73% so TBW = 75.5 x 73% = 55.1kg ; 55.1 / 97.9 = 56% 
That matches the scale result.

FFM = bone + water + muscle 
Checking that: 
51.7% (muscle from scale) of 75.5kg 
    + 3.9kg (bone from scale) 
    + 55.1kg (calculation above) = 97.9kg

If we can resolve either bone mass or muscle mass then we are done.

There seem to be a lot of variation on these formulas, though: Just look at this paper
http://www.scielo.br/pdf/rbcdh/v14n1/en_a03v14n1.pdf

dylanlt, 2018-04-19:
There's an ad-support app (yourscales.com) which does the calculation as well. It seems like he is using the Yunmai code library.

erijo, 2018-04-19:
What do you mean with "the Yunmai code library" @dylanlt?

oliexdev, 2018-04-28:
@dylanlt no, I haven't contacted Yunmai. I doubt that they will respond. Even if they have a library available it has to be compatible with the GPLv3.

erijo, 2018-05-01:
Similar to #159 there seems to be a native library (libconfig.so) at the bottom doing the conversion from resistance etc. to fat etc. If there is someone fluent in arm assembler or have an IDA license it might be possible to figure it out...

erijo, 2018-05-05:
You connect to the scale under settings/Bluetooth.

You can try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) if that works better.

Erol444, 2018-05-05:
Nevermind, at first the bluetooth icon on main screen didn't show up(a bug..?), now I just know where to click to turn it on. Just re-installed my yunmai app and tried to get it running so I could help you reverse engineer the rest of the info but like previously, I would have to download chinese version(no tnx)

[image]

Does the data from scale get sent to openScale so I could help(Bluetooth HCI Snoop Log) or would I need original yunmai app for that?

Erol444, 2018-05-05:
Ohh that's the problem. Saw few formulas up on this issue. So you are trying to figure out what formulas are used by the yunmai app so you can implement it in openScale? Wouldn't it be easier to just decompile the app itself and look into the source for that?
And this is the link that opens up if I click "confirm":
http://app.qq.com/#id=detail&appid=1102855655

erijo, 2018-05-05:
The problem is that the calculations aren't done in java, but in an included native library. I'm not as fluent in asm as I would need to be to figure that out :(

erijo, 2018-05-08:
Has anyone here with a Yunmai mini scale seen the issue in #261?

ivangill, 2018-06-07:
I tried the App "Your Scale" on the Yunmai.
Much better than yanmai app at finding scale
Most ibteresting, it calculates a noticeabley different figure for body fat %, and this new figure displays on the scales too. So the scales have a default calculation unless overrideen by the app...

oliexdev, 2018-09-21:
added water, muscle and bone mass for Yunmai mini with commit https://github.com/oliexdev/openScale/commit/2e8a2cdfedd36bb0809ac31916c0b01c96ec9c2c

closed 2018-09-21 by oliexdev

vitoresende, 2020-04-17:
Hi people, hi @oliexdev . Thanks for your help here. Did anyone pair / bond the app with yunmai? Is it possible?

Erol444, 2020-04-18:
> 
> 
> Hi people, hi @oliexdev . Thanks for your help here. Did anyone pair / bond the app with yunmai? Is it possible?

If you have a pro version than yes, it's straight forward.