oliexdev/openScale #74: Help to add lumsing SmartScale CF369BLE

chemstar, 2017-09-26:
Hi, following the wiki tutorial from this page i've re engineered some informations from my own bt scale.  so far i've discovered following:

**lumsing SmartScale CF369BLE**
http://en.lefu.cc/intelligent/22.html

**BT-Description:**
Name: Electronic Scale
UUIDs:  65520
             Custom Service: f000ffc0-0451-4000-b000-000000000000 ?
**BLE-Apps:**
Connection: fail
Type: LE only
Flags: LimitedDiscoverable
          BrEdrNotSupported
Incomplete List of 16-bis Service UUIDs:
  0xFFF0
  0000FFF0-0000-1000-8000-00805f9b34fb
**Reversed from Orig. Scale-App (some readable UUIDs):**
        "%08x-0000-1000-8000-00805f9b34fb"
 "00001101-0000-1000-8000-00805F9B34FB"
 "0000" + var2 + "-0000-1000-8000-00805f9b34fb"

**Wireshark:**
[localhost] 
_Write Request  		       0x0012
_Handle         			       0x0012
_Value          			       fe 01 01 01 b8 26 01 9e  (check EXP. A)
[scale]		
_Write Response 		       0x0013 
_Handle 		 		       0x0012
_Value			 	       no data
[localhost] 
_Write Request  		        0x0012
_Handle 		 		        0x0015 
_Value 		 		        01 00
[scale]		
_Write Response 			0x0013
_Handle 		 			0x0015
_Value			 		no data
[scale]		
_Handle Value Notification     0x1b
_Handle 		 			0x0014
_Value					cf 11 a6 b8 03 ad 00 b0 20 02 8c 0e 02 0d 07 64 (check EXP. B)
[scale] - 1xrepeated		
_Handle Value Notofication	0x1b
_Handle 		 			0x0014
_Value					cf 11 a6 b8 03 ad 00 b0 20 02 8c 0e 02 0d 07 64 (check EXP. B)

_EXP. A_
fe 01 01 01 b8 26 01 9e
fe - always set
01 - Ugroup	 ,P0-P9
01 - sex 		 ,00/01 	        women/men
01 - level 	 ,00/01/02 	ordinary/amateur/professional
b8 - height
26 - age
01 - unit		 ,01/02/04	kg/lb/st:lb/st 
9e - checksum 8 xor

_EXP. B_
cf 11 a6 b8 03 ad 00 b0 20 02 8c 0e 02 0d 07 64
      cf - sclae type 	        ,cf/cb/ce BODY_SCALE/BABY_SCALE/BATHROOM_SCALE
       1 - level
       1 - Ugroup
     a6 - sex/age combined (1010 0110, first bit from left sex, rest age 0x26)
     b8 - height			 (cm)
03 ad - weight			 (kg)
00 b0 - body fat		 (%)
     20 - bone			 (kg)
02 82 - muscle mass	 (kg)
     0e - visceral fat	 ()
02 0d - body water		 (%)
07 64 - bmr 			 (kcal) 

i hope the device can get be supported by the app.  thx

oliexdev, 2017-09-29:
Hi chemstar,

thanks for your reverse engineering work. I will implement that into openScale but I need a bit more information.

Can you write down the service and uuid which is EXPA and EXPB was send/receive to? You can see it in wireshark if you collapse the "Handle: xxx" item or upload the log file so I can see by myself.

Which Bluetooth device name and device hardware address has the scale?

Can you send me a photo of the scale at the same angle as in the wiki? 

Best regards
oli

oliexdev, 2017-09-29:
I implemented a first try with commit https://github.com/oliexdev/openScale/commit/a98cf937dd701c51e1504e9d208ebe99a138e578
But I need more information to get it work, so please post your log file.

Thank you in advance

chemstar, 2017-09-29:
log attached
for further informations you can take a look at iwellness4.0 app (https://play.google.com/store/apps/details?id=com.lefu.iwellness4.system&hl=de) using show java decompiler (https://play.google.com/store/apps/details?id=com.njlabs.showjava&hl=de)

[btsnoop_hci.zip](https://github.com/oliexdev/openScale/files/1345655/btsnoop_hci.zip)

Regards

oliexdev, 2017-09-30:
thanks @chemstar but sadly somehow in the handle they are no UUIDs...
Does the [BLE scanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner) works? FYI sometimes it needs several times that the connection is established to the scale.

chemstar, 2017-09-30:
Yes i have seen that. It take some tries to get the scale data even with the iwellness app.

oliexdev, 2017-10-01:
then it will be really hard for me to implement it without using the scale because I have to try and test out which UUID and Bluetooth characteristic are used. Currently I take your UUID from above. 

Maybe I guessed right. You can test the latest developer version if you want.

Does the developer build connect to the scale?
If yes, do you receive any data?
If yes, are the weight and other values correct?

chemstar, 2017-10-02:
unfortunately it doesnt work, 
**but** following some examples on the net i could make some proges. based on sample ble apps and informations from (https://www.allaboutcircuits.com/projects/how-to-communicate-with-a-custom-ble-using-an-android-app/)  i could establishing connection and read some data from it.
the adaptations i've made:

    public void readCustomCharacteristic() {
        if (mBluetoothAdapter == null || mBluetoothGatt == null) {
            Log.w(TAG, "BluetoothAdapter not initialized");
            return;
        }
        /*check if the service is available on the device*/
        BluetoothGattService mCustomService = mBluetoothGatt.getService(UUID.fromString("0000FFF0-0000-1000-8000-00805f9b34fb"));
        if (mCustomService == null) {
            Log.w(TAG, "Custom BLE Service not found");
            return;
        }
        /*get the read characteristic from the service*/
        BluetoothGattCharacteristic mReadCharacteristic = mCustomService.getCharacteristic(UUID.fromString("0000FFF4-0000-1000-8000-00805f9b34fb"));
        setMyCharacteristicNotification(mReadCharacteristic, true);
        if (mBluetoothGatt.setCharacteristicNotification(mReadCharacteristic, true)) {
            Log.w(TAG, "Failed to set notify characteristic");
        }
    }

    public void writeCustomCharacteristic(byte[] data) {
        if (mBluetoothAdapter == null || mBluetoothGatt == null) {
            Log.w(TAG, "BluetoothAdapter not initialized");
            return;
        }
        /*check if the service is available on the device*/
        BluetoothGattService mCustomService = mBluetoothGatt.getService(UUID.fromString("0000FFF0-0000-1000-8000-00805f9b34fb"));
        if (mCustomService == null) {
            Log.w(TAG, "Custom BLE Service not found");
            return;
        }

        /*get the read characteristic from the service*/
        BluetoothGattCharacteristic mWriteCharacteristic = mCustomService.getCharacteristic(UUID.fromString("0000FFF1-0000-1000-8000-00805f9b34fb"));
        mWriteCharacteristic.setValue(data);
        if (mBluetoothGatt.writeCharacteristic(mWriteCharacteristic) == false) {
            Log.w(TAG, "Failed to write characteristic");
        }
    }

    public void setMyCharacteristicNotification(BluetoothGattCharacteristic bluetoothGattCharacteristic, boolean bl) {
        if (this.mBluetoothAdapter == null || this.mBluetoothGatt == null) {
            Log.w((String)TAG, (String)"BluetoothAdapter not initialized");
            return;
        } else {
            this.mBluetoothGatt.setCharacteristicNotification(bluetoothGattCharacteristic, bl);
            BluetoothGattDescriptor bluetoothGattDescriptor = bluetoothGattCharacteristic.getDescriptor(UUID.fromString((String)SampleGattAttributes.CLIENT_CHARACTERISTIC_CONFIG));
            if (bluetoothGattDescriptor == null) return;
            {
                bluetoothGattDescriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                this.mBluetoothGatt.writeDescriptor(bluetoothGattDescriptor);
                return;
            }
        }
    }

[new_log.zip](https://github.com/oliexdev/openScale/files/1348922/new_log.zip)

oliexdev, 2017-10-02:
I updated the UUIDs with commit https://github.com/oliexdev/openScale/commit/777c84dd3b4cb0cabeadb68ea75906a0c5d319fe

Please test the [developer build version 69](https://github.com/oliexdev/openScale/blob/master/openScale-dev-build.apk).

chemstar, 2017-10-02:
i've made some progress and i think i'm more than close to the happy end ;-)
- Method: nextInitCmd(..)
case 1: causes an Unhandled exception in callback ... BluetoothGattDescriptor.setValue(byte[])' on a null object reference
- Method: nextBluetoothCmd(...)
case 1: 
setNotificationOn(WEIGHT_MEASUREMENT_SERVICE, WEIGHT_CUSTOM0_CHARACTERISTIC, WEIGHT_MEASUREMENT_CONFIG);
- Method: onBluetoothDataChange
comparisson should be:
        if (data.length == 32 && data[0] == (byte)0xCF) {
- Method: parseBytes
add missing:
addScaleData(scaleBtData);

oliexdev, 2017-10-03:
> i've made some progress and i think i'm more than close to the happy end ;-)

great :)

> case 1: causes an Unhandled exception in callback ... BluetoothGattDescriptor.setValue(byte[])' on a null object reference

Ok maybe the UUID `0000FFF1-0000-1000-8000-00805f9b34fb` don't have a notification flag. I deleted this state.

> setNotificationOn(WEIGHT_MEASUREMENT_SERVICE, WEIGHT_CUSTOM0_CHARACTERISTIC, WEIGHT_MEASUREMENT_CONFIG);

ok moved to the `nextBluetoothCmd(...)`

> if (data.length == 16 && data[0] == (byte)0xCF) {

true

> add missing:
> addScaleData(scaleBtData);

addScaleData is already at the end of parseByte method.

> additional new data dialog would be usefull asking the user for commiting new data

you can already add new data. Just click on the plus button on the bottom in the overview tab.

> Support for additional informations (check iwellness4 app) like BMI scale, bone weight, bmr, visceral fat, user levels (if even possible). i can give support if needed.

BMI, bone weight, BMR are already supported. What do you mean with user levels?

I made the modifications with commit https://github.com/oliexdev/openScale/commit/1d89901eebece8887adb427a9a9daa75980f2925, so check out the new developer build version.

chemstar, 2017-10-03:
thx oli,

> additional new data dialog would be usefull asking the user for commiting new data
>>you can already add new data. Just click on the plus button on the bottom in the overview tab.

i didn't mean the manual input, but input over bt. 
on enabled "smartUserAssign" and selected user"A" the meassurement for user"B" will be added to the correct profile but without any visible feedback.

>Support for additional informations (check iwellness4 app) like BMI scale, bone weight, bmr, visceral fat, user levels (if even possible). i can give support if needed.
>>BMI, bone weight, BMR are already supported. What do you mean with user levels?

in the scale app i got an option to set different user levels:
01 - level ,00/01/02 ordinary/amateur/professional (fe 01 01 "01" (<- this one) b8 26 01 9e)

**explaination:**
* Ordinary: people who do exercise less than or equal to two days per week
* Amatheur athletes: people who do professional exercise lass than or equal to 5 days per week
* Professional athletes: people who do professional exercise evry day.
Depending on this param scale does the calculation. 

**additional:**
- gui (table view):
table view shows the last data entry even when the last entry is deleted (refresh needed). diagramm/stats doesn't show anything about this entry.

- BluetoothCommunication.startSearching(...)
shouldn't be better to call stopLeScan() before device.connectGatt(context, false, gattCallback)

oliexdev, 2017-10-03:
> i didn't mean the manual input, but input over bt.
> on enabled "smartUserAssign" and selected user"A" the meassurement for user"B" will be added to the correct profile but without any visible feedback.

ok I implemented with commit https://github.com/oliexdev/openScale/commit/b0fe50c365fe047c8fb53eaf6bbd3a964dd7063a a toast text which shows the weight, the date and to which user it was added.

> in the scale app i got an option to set different user levels:
01 - level ,00/01/02 ordinary/amateur/professional (fe 01 01 "01" (<- this one) b8 26 01 9e)

ok I understand but that would take more time to implement.

> 
> gui (table view):
> table view shows the last data entry even when the last entry is deleted (refresh needed). diagramm/stats doesn't show anything about this entry.

thanks it was a bug it should be fixed now

> BluetoothCommunication.startSearching(...)
> shouldn't be better to call stopLeScan() before device.connectGatt(context, false, gattCallback)

stopSearching is already be called before startSearching in the MainActivity class

But you didn't tell if your scale now works with openScale?

oliexdev, 2017-10-03:
only one data transmission of the same content is allowed with commit https://github.com/oliexdev/openScale/commit/77e3a4c7d1fbce91c2b5ed5be95b3171744fa849

and "case 2" is removed.

chemstar, 2017-10-03:
app is working, but sometimes it need some attempts to connect to device. i thought that:
> BluetoothCommunication.startSearching(...)
> shouldn't be better to call stopLeScan() before device.connectGatt(context, false, gattCallback)
>> stopSearching is already be called before startSearching in the MainActivity class
will change something. requires some further investigation.

- an issue in onBluetoothDataChange(...)
use "if (!Arrays.equals(data, receivedData)) { // accepts only one data of the same content" for doing the comparisson

- another question:
sometime the scale can't calculate other values so only the weight is transmited. how are this missing values evaluated inside app-diagram? is there a gap for this missing values or does the curve fall from last valid data to "0"?

oliexdev, 2017-10-04:
ok changed it.

> another question:
> sometime the scale can't calculate other values so only the weight is transmited. how are this missing values evaluated inside app-diagram? is there a gap for this missing values or does the curve fall from last valid data to "0"?

there is not really a gap and is not falling to 0. There is no datapoint for the missing values but if the next measurement has values again then the diagram is connected to the last values.

I assume the scale is working now, so I close this issue.

Can you please provide a photo of your scale for the openScale wiki site? (same angle like the other photo)

closed 2017-10-04 by oliexdev

chemstar, 2017-10-04:
only two things more,

>additional codes:
>_ [fd 35 00 00 00 00 00 35], [NotifOn(WEIGHT_CUSTOM0_CHARACTERISTIC)]
>-> scale power off

shouldn't it be used for nextCleanUpCmd after successfull data transfer?

- and due to completeness, some additional error-codes:
[fd 31 00 00 00 00 00 31] - ERROR_CODE
[fd 33 00 00 00 00 00 33] - ERROR_CODE_GETDATE
[fd 31 00 00 00 00 00 33] - ERROR_CODE_TEST

i will take a photo and attach it here

oliexdev, 2017-10-06:
thanks @chemstar yeah if you can implement the stop command and test it maybe?

chemstar, 2017-10-07:
- add  "android.permission.ACCESS_COARSE_LOCATION" to AndroidManifest.xml
(due to: https://stackoverflow.com/questions/33045581/location-needs-to-be-enabled-for-bluetooth-low-energy-scanning-on-android-6-0)
- i've tested the poweroff cmd, but i see no sense of the function because the device will turn off itself  after short time of inactivity.

```java   
   @Override
    boolean nextCleanUpCmd(int stateNr) {
        switch (stateNr) {
            case 0:
                // send poweroff command to scale
                byte[] poweroffCmd = {(byte)(0xfd), (byte)(0x35), (byte)(0x00), (byte)(0x00), (byte)(0x00), (byte)(0x00), (byte)(0x00), (byte)(0x35)};
                writeBytes(WEIGHT_MEASUREMENT_SERVICE, WEIGHT_MEASUREMENT_CHARACTERISTIC, poweroffCmd);
                break;
            case 1:
                setNotificationOn(WEIGHT_MEASUREMENT_SERVICE, WEIGHT_CUSTOM0_CHARACTERISTIC, WEIGHT_MEASUREMENT_CONFIG);
                break;
            default:
                return false;
        }

        return true;
    }
````
- some additional informations from the manual
(same as https://blueweigh.com/bmi-smart-scale-tutorial/)

SPEC's:
...
02. Unit: kg/lb
03. 10 user groups
...
05. Three fitness levels: Ordinary, Amateur and Professional
06. Capacity: 180kg/400lb
07. Minimum, weight: 5kg/11 lb
08. Muscle Mass measurement graduation: 0.1kg
09. Weight measurement graduation: 0.1kg/0.2 lb
10. Body Fat measurement graduation: 0.1%
11. Body Water measurement graduation: 0.1%
12. Bone graduation: 0.1 kg
13. Age range: 10-100 years old
14. Height range: 100-255cm 3ft:3in-8ft:4in
...

FAQs:
....
2. Display indicates”Err” when weighing:
– Overload the maximum range (182.5kg/405.5lb)
...
3. Display indicates”Err2″ when measuring body fat:
– Body parameter settings should be with:
Height 100cm-255cm, 3ft: 3in-8ft: 4in, Aged: 10-100 years old.
– Underweight less than 30kg/66lb.
– Feet or scale are too wet.
4. No body fat data measured when weighing.
– Body fat scale requires weighing with bare feet.
– Body data setting should be with: Height 100cm-255cm (3ft: 3in-8ft: 4in), Age 10-100 years old.
– Bluetooth function not open.
– Paired with other connected electronic scales.

**and thank you for your fantastic work and support!**

oliexdev, 2018-08-31:
@darkcheater looks more like you have a 1byone scale, see issue #159 for how the progressing is going on.

darkcheater, 2018-09-01:
Thanks for your fast answer.
OK, strange. How are the two models differentiated? 
Here are two more pictures that show why I thought it should be CF369BLE scale.. 
[image]
[image]

PenguinIsrael, 2025-03-25:
I have cf369ble but app doesnt get data to show