oliexdev/openScale #77: Medisana BS 430 doesn't work

baztian, 2017-10-02:
Hi, I'm having a BS 430. Unfortunately it isn't recognized when I hit the Bluetooth button.
Am I doing something wrong? Is it simply unsupported. Can I help somehow reporting some technical data or analyzing with hcitool?
Thanks

oliexdev, 2017-10-03:
Hi baztian,

No you didn't do something wrong. The Medisana BS 430 is currently not supported only the Medisana BS444 one.
Sure you can help. You could try to reverse engineer your scale, see the openScale wiki for more information. Maybe they use also the same protocol like the BS444...

Best regards
oli

baztian, 2017-10-03:
Hi oli, I was hoping that BS 430 is compatible with BS 444. According to [keptenkurk's BS440 Project](/keptenkurk/BS440) that software supports both and they name both it in one breath. In the code I didn't find a differentiated handling.
What if the protocol is the same? What needs to be changed to openScale to support BS 430? Why didn't it detect the BS 430?
Anyway, I'll have a look at the reverse engineering Wiki page and'll at least try to supply some information.

Thanks

oliexdev, 2017-10-03:
Could be possible that BS 430 is compatible with BS 444.
You have to change in the [BluetoothMedisanaBS444 class](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothMedisanaBS444.java) the method `checkDeviceName(String btDeviceName)` so that is accepts the BS 430 device name.

baztian, 2017-10-03:
How can I find out the btDeviceName? Using linux btmon when doing a hcitool lescan I get

    Name (complete): 0203B 02E6B64754C2

That's also the information the BLE Scanner app gives me. Also under DEVICE NAME I get a UUID `00002A00-0000-1000-8000-00805F9B34FB`.

For what String shall I search in the `checkDeviceName` method?

Thanks

oliexdev, 2017-10-04:
Medisana is using weird device names. The last digits is your individual Bluetooth hardware address.
Check for the beginning of the device name. Use this code snippet:

```
    @Override
    public boolean checkDeviceName(String btDeviceName) {
        if (btDeviceName.startsWith("013197") || btDeviceName.startsWith("0203B")) {
            return true;
        }

        return false;
}
```

Please give feedback if it works.

yacc143, 2017-10-05:
I've got a Medisana BS444 (it says so on the back side, and it looks exactly like the photo in the wiki), but it has a completely different HW ADDR:

name: 0805A012430ADD addr: D4:DE:29:05:FB:19

Guess I need to brush up how to compile apk files after patching the detection functionality to detect it?

[image]

yacc143, 2017-10-05:
Any idea how this is supposed to work?

I've used a patched apk so my Medisana BS 444 was recognized as such in settings/bluetooth/device.

But I tried to hit the Bluetooth button on the top right before/during/after using the scale, and I managed a couple times to get the connection dots on both sides of the bluetooth button. No data.

And the + Button at the bottom seems to be a manual data entry function?

TiA, Andreas

baztian, 2017-10-06:
@oliexdev I patched the java file and built it using travis as I didn't want to setup a android build environment locally. I tweaked the travis file to upload as [github release](/baztian/openScale/releases). When I try to install the file Android asks me to do so but then it says "App wurde nicht installiet" (App wasn't installed). I've allowed for insecure origins.
Any ideas?

baztian, 2017-10-06:
@oliexdev nevermind. I had to uninstall the app first. Now it starts normally.

baztian, 2017-10-06:
It says "connecting to 0203B 02E6B64754C2". But after some seconds it says "disconnected". No measurements were recorded.
I guess I have to really work through the wiki.

oliexdev, 2017-10-06:
@yacc143 does your scale works with the original app?
While you are using a patched apk file you should receive some data... Maybe you can check the used UUIDs from your scale with the UUIDs used by openScale?

@baztian maybe they just change a little thing?

yacc143, 2017-10-06:
@oliexdev Just assuming, this should work, right?

root@miraculix:~# sudo service bluetooth restart
root@miraculix:~# hcitool leinfo D4:DE:29:05:FB:19
Requesting information ...
Could not create connection: Connection timed out
root@miraculix:~# hcitool leinfo D4:DE:29:05:FB:19
Requesting information ...
Could not create connection: Input/output error

First time the leinfo takes a long time (couple of seconds at least) before it times out, second time it's immediate an IO error. (I've tried it with the scale display active while putting down a foot on it)

oliexdev, 2017-10-07:
@yacc143 It looks like you bought a faulty hardware. I never used hcitool I mostly use the BLE scanner App by Bluepixel Technology LLP.

oliexdev, 2017-10-17:
I will close this issue now if anyone has more information or a solution please create a new issue or create a pull request.

closed 2017-10-17 by oliexdev

vershinin, 2018-12-15:
BS 430 has the same protocol as BS 444, however timestamp offset is not required there.
I had prepared PR #363 with patch for BS 430 support.

baztian, 2019-01-03:
Unfortunately bs 430 still doesn't work for me. It says connected and then two seconds later connection lost. Any ideas?

vershinin, 2019-01-03:
For me it was tricky too even get it working with stock Medisana software. However, I figured out that problem was BLE implementation in Huawei firmware, which is quite buggy. I restarted my smartphone, and it started to work, but very unstable. I had finished with Esp32 module, which listens for BLE data and uploads it to Google Fit cloud, but it's another story.