oliexdev/openScale #81: Improve measurements entry

erijo, 2017-10-11:
On my scale, after measurements have been taken, the values for body fat, muscles etc. loop by twice after which they disappears. To make measurements entry in the app smoother it would be nice if you could configure the order in which the measurements appear during entry to be the same as on the scale. Then after entering the weight and pressing ok (or next) it would display the next entry in the configured order. It would also be nice if there where a + and - button on the entry screen to quickly increase or decrease a value by e.g. 0.1 kg.

oliexdev, 2017-10-17:
Hi erijo,

I understand your problem and yeah your idea sounds good but to configure the order is not simple to implement yet. But I will keep that in mind. Thanks.
The + and - buttons is much easier to implement. If I have a bit more time I will implement it.

oliexdev, 2017-10-22:
added + and - buttons with commit https://github.com/oliexdev/openScale/commit/22d812a2ccd97bfbd407997959fbb4ed20532804

closed 2017-10-22 by oliexdev