oliexdev/openScale #82: feature request. MQTT support

KavajNaruj, 2017-10-16:
Hi, first i want to thank you for this great app, definitely better than original [Health Coach](https://play.google.com/store/apps/details?id=de.sanitas_online.healthcoach&hl=en)

I'd like to ask if is possible to add support to publish data after measurement to [MQTT broker](https://github.com/mqtt/mqtt.github.io)? I am not fan of regularly exporting. In addition to backups it can add other uses such as TTS to the BT speaker, or send data to Influx and visualize them in Grafana

oliexdev, 2017-10-17:
I am not a big fan to send health data to any public clouds for privacy reasons (one reason why I started openScale, see [here](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale#how-its-started-)) but of course you are free to do it. 
If you want you can export the data and write a script to send it via MQTT protocol. Another solution  is to write for e.g. lets call it "openScale Connector" app which listen and access the internal sql database of openScale and send any new data via MQTT or to any company clouds (GARMIN,Google Fit and so on). I am willing to maintain and help this new "connector app" but the cloud implementations must come from the open source community.

KavajNaruj, 2017-10-18:
No clouds! I am running MQTT broker on LAN and I'd like to have a backup on NAS locally .

The MQTT is much better than SQL. If you subscribe for "topic" like lets say "openScale/user/weigth" you get values in real time, instead of repetable asking SQL database for changes.

So that kind of application that you mention (GARMIN,Google Fit and so on) will be esier to write, but that is not my consern.

Acually my plan is to connect openScale to my [HomeAssistant](https://home-assistant.io) instance and use it for some cool automation, like get praise when I lose some weigth an so on.

I think the best solution for this is to put some MQTT config ti openScale settings like:
- Host
- Port
- User
- Password

And after measurement send data to MQTT topics trougth [MQTT client](https://github.com/eclipse/paho.mqtt.android)

oliexdev, 2017-10-22:
ok send a PR with your implementation. Either I will create a connector app as describe above or I will integrate it directly into openScale.

closed 2017-10-22 by oliexdev