oliexdev/openScale #89: java.lang.NumberFormatException: Invalid int: "java.lang.NumberFormatException: Invalid int: "300416523230576771213457683014051600570377573""

tianxiaogu, 2017-11-02:
The parseInt at [UserSettingsActivity:#261](https://github.com/oliexdev/openScale/blob/8f7600f5fffa12cca1d0ba4d8177c0fced24f7f0/android_app/app/src/main/java/com/health/openscale/gui/activities/UserSettingsActivity.java#) should handle very large interger.

```
// CRASH: com.health.openscale (pid 6003) (elapsed nanos: 5771939483589)
// Short Msg: java.lang.NumberFormatException
// Long Msg: java.lang.NumberFormatException: Invalid int: "300416523230576771213457683014051600570377573"
// Build Label: Android/sdk_google_phone_x86/generic_x86:6.0/MASTER/4088240:userdebug/test-keys
// Build Changelist: 4088240
// Build Time: 1497047463000
// java.lang.NumberFormatException: Invalid int: "300416523230576771213457683014051600570377573"
// 	at java.lang.Integer.invalidInt(Integer.java:138)
// 	at java.lang.Integer.parse(Integer.java:413)
// 	at java.lang.Integer.parseInt(Integer.java:367)
// 	at java.lang.Integer.parseInt(Integer.java:334)
// 	at java.lang.Integer.valueOf(Integer.java:525)
// 	at com.health.openscale.gui.activities.UserSettingsActivity$onClickListenerOk.onClick(UserSettingsActivity.java:261)
// 	at android.view.View.performClick(View.java:5198)
// 	at android.view.View$PerformClick.run(View.java:21147)
// 	at android.os.Handler.handleCallback(Handler.java:739)
// 	at android.os.Handler.dispatchMessage(Handler.java:95)
// 	at android.os.Looper.loop(Looper.java:148)
// 	at android.app.ActivityThread.main(ActivityThread.java:5417)
// 	at java.lang.reflect.Method.invoke(Native Method)
// 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
// 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
//
```

oliexdev, 2017-11-08:
How did you end up with such a great number for the user height?

Anyway thanks I fixed it with commit https://github.com/oliexdev/openScale/commit/481eb17d367f5d1c5012c5caed88708d145c9cf9

closed 2017-11-08 by oliexdev

tianxiaogu, 2017-11-08:
Hi @oliexdev, thank you for taking time to fix this issue. I was evaluating my automated GUI testing tool on a set of well-maintained open source apps, and this app is one of them.

oliexdev, 2017-11-08:
ok I understand :wink: 
well then you will definitely find more bugs in openScale because I didn't really catch all user mistakes...