oliexdev/openScale #91: [FR] Dark theme

woj-tek, 2017-11-05:
It would be cool if the application had a dark/black theme.

oliexdev, 2017-11-08:
good idea but need to figure out how to implement that. I don't really like the xml stuff in android... :/
so any help are welcome.

oliexdev, 2018-01-28:
done with commit https://github.com/oliexdev/openScale/commit/2dc40e4bc45adcfd867b4f9e14ac60e4da2c662a

closed 2018-01-28 by oliexdev

erijo, 2018-01-28:
Nice work @oliexdev, looks really nice!

One thing I noticed is that the text color and indicator color in the gauge should probably not be black.

oliexdev, 2018-01-28:
I missed that one. It is fixed now. :smile:

erijo, 2018-01-29:
One more thing: should the text color really be gray for the light theme? It doesn't look as good as it did before when the text was black (IMHO).