oliexdev/openScale #92: Bodyfat etc. on Digoo DG-S038H

dynobo, 2017-11-08:
Hi,
I experience some trouble with the Measurement of Bodyfat and the other Health Date: Sometimes (~every 5-10th measurement?) they are calculated an transfered to OpenScale correctly, but most of the time, they are missing. Bodyweight however works perfectly and get's shown every time.
Any hints? :)

oliexdev, 2017-11-08:
Hi dynobo,

I can't really help you because I don't own this scale @Murgeye in the PR #73 reverse engineered and implemented this scale.

Best regards
oli

dynobo, 2017-11-10:
Thanks, I had also other issues and the Scales quality doesn't convince me either. I'll send it back, so this is not relevant for me anymore.

closed 2017-11-10 by dynobo

Murgeye, 2017-11-10:
Just for reference: All data except the weight is transferred after a short delay. If you don't stand on it long enough they won't show up. Just wait for the weight display on the scale to stop blinking (no idea why it takes so long to calculate, the original app shows the same behavior for me).

dynobo, 2017-11-10:
That wasn't the case for me: I stand on the scale, the light was blinking, bluetooth connected, and I stepped off after the scales display (and the scale itself?) switched off. However, I also was under the impression that the timing (between switching on, connecting to bluetooth and measurement) somehow had an influence here. Just couldnt figure out a pattern...