oliexdev/openScale #93: Initial setup with Digoo DG-S038H

dynobo, 2017-11-08:
Hi, just wondering:
Initially I wasn't able to connect with the Scale: Bluetooth connection couldn't be established.
I downloaded the vendors App to test the Hardware, which was working fine. After that, also OpenScale was able to connect perfectly.

Later, I removed the vendors app and the connection with OpenScale keeps working.

Is this the intended/usual behavior? That the venders App is needed for initial setup? If yes, it would be nice to have it in the Readme.

PS: I'm big fan of OpenScale, used it for a certain time doing manual entries, and recently bought the Digoo especially to use it with OpenScale. If any testing with this setup is needed, I would be glad to help :)

oliexdev, 2017-11-08:
Yeah must of the scales needs to be initialized for example to set the time to enable the body measurement functions and so on. I guess with your scale the initial process was not reverse engineered and implemented into openScale. So yeah you can help to do it. You need to record the initial process via Bluetooth with the vendor app. How you do it please read the openScale wiki. If you have any questions just ask.

dynobo, 2017-11-10:
Thanks for your comment, but as I also have other issues with the scale, i'm going to send it back and can't help here. I'm leave it to you, to close the issue or wait for someone else to work on it.

Murgeye, 2017-11-10:
I guess I'll spend some time to find out who to reset the initialization process for my scale when I find the time. Sorry, I didn't realize that part was missing because it just works for me :smile:

oliexdev, 2017-11-19:
I will close this issue for now.
@Murgeye would be great if you find some time to look into it.

closed 2017-11-19 by oliexdev