oliexdev/openScale #96: Good scale to use with OpenScale?

dynobo, 2017-11-10:
Hi,

What are your experience with Scales connected to OpenScale? Which Scale is supported good or best?

Background: I tried the Digoo Scale, but had some issues with it in combination with OpenScale: Initial pairing had to be done with the vendors app and no other values than weight were transferred. I'll send it back and look for an alternative...

Thanks!

oliexdev, 2017-11-10:
As I don't own all scales by myself I can't really say which works the best with openScale.

But here are few thoughts which you may consider in your decision:

- The best control of the scale is to hack a scale like the [Custom Bluetooth scale](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale) which was the starting point of openScale. You have full control of everything and it worked great. Of course the disadvantage is that this variant is the most complicated one and I did the connection with Bluetooth 3.x. Nowadays Bluetooth 4.x is standard and that is sadly not downwards compatible. 

-  If you only want the weight I recommend using Xiaomi Mi scale in the version 1. I would say it works good with openScale. But I don't know what happens when the internal RAM of the scale is full!? Because no scale data will ever be deleted. The advantage is the you don't have to pair with openScale every time you weight. Just sync next time with openScale and you get all last weight data.

- If you want also measure your body fat and so on I would recommend using Sanitas SBF70. Note: I don't own this scale but as I hear from the users most are satisfied with openScale with that scale. Here is also the question what happened when the internal RAM capacity is reached...!? The advantage is that the body fat calculation and so on is done on the scale itself via impedance measurement.

- Yunmai mini and Xiaomi Mi Scale version 2 are currently only basic supported because the body water calculation and so on are done in the App and not on the scale. Only the impedance value are transmitted (beside Yunmai mini the fat percentage is also transmitted). How they calculate the rest of the body metrics like bone mass and muscle percentage are currently not known.

- But as I discovered most body metrics are mathematical derived from the measured weight and the fat percentage measured via the [bioelectrical impedance analysis](https://en.wikipedia.org/wiki/Bioelectrical_impedance_analysis). This method is general less accurate if you don't use a clinical device. As above describe how the companies are calculating the body metrics are not transparent and are not public published e.g. in a scientific paper.

- In the development version of openScale (in the next release) I Implemented body metrics estimations for the body fat, water percentage and body lean weight regarding to published scientific papers. But these are estimations and no impedance values are used for the calculations! I will link these papers in the wiki which I used so everybody can see which calculation they want to use.

As you see a overall perfect scale with perfect measurements doesn't exist for the home use. 
But to make it more transparent I will update the wiki with a FAQ and for each scale a short entry which functions are work and which may not. 

I hope I could answer your question a little bit.

dynobo, 2017-11-11:
Thanks, that helps me a lot :-)

And yes, I think it would be a good idea, to create a place in the Wiki, where experiences with the different scales and OpenScale can be shared. 👍

oliexdev, 2017-11-19:
added an overview page of supported scales at https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale

closed 2017-11-19 by oliexdev

MarvinT, 2018-01-20:
Are the impedence values logged in the app for the xiaomi scale?

Peterkn2001, 2020-01-29:
Hi, I have a slightly unusual project that I would like some advice on, if you don't mind...

My requirement is to weigh domestic bottled gas cylinders to know roughly how much gas is left in them. The gas bottles will sit on the scales all of the time, and I'd like to obtain weight data fairly infrequently (every hour maybe, possibly once per day). I have 4 bottles in total.

My initial intention was to build my own scales and use an ESP8266 with wifi connection, and implement deep sleep to conserve battery life. The main issue with this approach would be the mechanical engineering aspect, especially as I need 4 scales that work in the same way. I'd considered hacking several Wii Fit boards and fitting my own electronics in the cases - and this may still be the best approach - but I like the simplicity of using a ready made off the shelf smart scale instead.

My main question is about whether there is a brand of smart scale that works well with a constantly applied load and which will push-out the current weight reading every so often?
I assume that most scales are triggered when a person steps on to them, and they wake-up and start transmitting Bluetooth data for a while then shut down again. I obviously need something different to this to achieve what I'm looking for.
Any ideas?

Thanks in advance.
Pete.

bluesdataventure, 2021-12-07:
> As I don't own all scales by myself I can't really say which works the best with openScale.
> 
> But here are few thoughts which you may consider in your decision:
> 
>     * The best control of the scale is to hack a scale like the [Custom Bluetooth scale](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale) which was the starting point of openScale. You have full control of everything and it worked great. Of course the disadvantage is that this variant is the most complicated one and I did the connection with Bluetooth 3.x. Nowadays Bluetooth 4.x is standard and that is sadly not downwards compatible.
> 
>     * If you only want the weight I recommend using Xiaomi Mi scale in the version 1. I would say it works good with openScale. But I don't know what happens when the internal RAM of the scale is full!? Because no scale data will ever be deleted. The advantage is the you don't have to pair with openScale every time you weight. Just sync next time with openScale and you get all last weight data.
> 
>     * If you want also measure your body fat and so on I would recommend using Sanitas SBF70. Note: I don't own this scale but as I hear from the users most are satisfied with openScale with that scale. Here is also the question what happened when the internal RAM capacity is reached...!? The advantage is that the body fat calculation and so on is done on the scale itself via impedance measurement.
> 
>     * Yunmai mini and Xiaomi Mi Scale version 2 are currently only basic supported because the body water calculation and so on are done in the App and not on the scale. Only the impedance value are transmitted (beside Yunmai mini the fat percentage is also transmitted). How they calculate the rest of the body metrics like bone mass and muscle percentage are currently not known.
> 
>     * But as I discovered most body metrics are mathematical derived from the measured weight and the fat percentage measured via the [bioelectrical impedance analysis](https://en.wikipedia.org/wiki/Bioelectrical_impedance_analysis). This method is general less accurate if you don't use a clinical device. As above described how the companies are calculating the body metrics are not transparent and are not public published e.g. in a scientific paper.
> 
>     * In the development version of openScale (in the next release) I Implemented body metrics estimations for the body fat, water percentage and body lean weight regarding to published scientific papers. But these are estimations based on your weight, age, sex, height but no impedance values are used for the calculations! I will link these papers in the wiki which I used so everybody can see which calculation they want to use.
> 
> 
> As you see a overall perfect scale with perfect measurements doesn't exist for the home use. But to make it more transparent I will update the wiki with a FAQ and for each scale a short entry which functions are work and which may not.
> 
> I hope I could answer your question a little bit.

Hello! 

As it has been 4 years since you made these recommendations do they still hold up? Which would you recommend today? 

Unfortunately, the  Sanitas SBF70 is no longer available in the USA.

bbigras, 2022-07-27:
Also since then, did people figured out if the "internal RAM capacity is reached" thing was a problem?

daleyjem, 2023-02-07:
> Hi, I have a slightly unusual project that I would like some advice on, if you don't mind...
>
> My requirement is to weigh domestic bottled gas cylinders to know roughly how much gas is left in them. [...]

@Peterkn2001 
I've had this exact same application need, only with water cooler jugs and beer kegs. Did you ever find an answer? I also went down the route of trying to build my own scale, even 3D modeled a bunch of parts, but eventually abandoned the project. My scale accuracy just wasn't great. If I could find a way to supply constant power for doing periodic or constant weight monitoring, that would be nice.

Peterkn2001, 2023-02-07:
@daleyjem no, never took it any further.
I do have plans to build a weighing system for my gas bottles using strain gauges and a battery-powered ESP8266 using deep sleep to send occasional weight updates via MQTT, but haven’t made a start on it yet.

tojo99, 2023-12-29:
> Hi, I have a slightly unusual project that I would like some advice on, if you don't mind...
>
> My requirement is to weigh domestic bottled gas cylinders to know roughly how much gas is left in them. [...]

Hi, you found a solution, this is exactly what interests me too.

Peterkn2001, 2023-12-29:
> Hi, you found a solution, this is exactly what interests me too.

@tojo99 I did find a solution, but it involved me taking a regular digital scale (I used ones by Grundig) and replacing the circuit board with one of my own which uses a Seeed Studio XIAO ESP32C3 and a small rechargeable battery, which is charged via a small solar panel (one panel charges two scales which sit side-by-side in a storage box). I use deepsleep to wake the controllers, take a reading and connect to WiFi and MQTT to send the reading.

tojo99, 2023-12-29:
> > Hi, you found a solution, this is exactly what interests me too.
> 
> @tojo99 I did find a solution, but it involved me taking a regular digital scale (I used ones by Grundig) and replacing the circuit board with one of my own which uses a Seeed Studio XIAO ESP32C3 and a small rechargeable battery, which is charged via a small solar panel (one panel charges two scales which sit side-by-side in a storage box). I use deepsleep to wake the controllers, take a reading and connect to WiFi and MQTT to send the reading. The hardware is in a different country at the moment, and I don't have any photos at the moment, so that's about as much info as I can give for now.

@Peterkn2001 Thanks for the answer, but it's too complicated for me.