oliexdev/openScale #97: Medisana BS444 not showing up in BLE scanner

wasbeer, 2017-11-18:
Hi,

my Medisana BS444 scale does not show up in the built-in BLE scanner. Is this normal? 

It also doesn't show up in the third party BLE scanner app. I did not register the scale yet with Vitadock.

Is it still necessary to add the MAC to the class, or does the BLE scanner cover this part?

oliexdev, 2017-11-19:
No you don't have to add a MAC address the built-in BLE scanner in openScale search for your device name. 
When your scale is on your device should be found by a BLE scanner normally.
Does it work with the original app? Maybe you bought a faulty scale.

oliexdev, 2017-11-25:
I will close this issue. Comment here if you have any further questions.

closed 2017-11-25 by oliexdev