oliexdev/openScale #99: French translations adjustments

fabads, 2017-11-19:
Hello,

Below some suggestions to improve the french translations:
- For harmonization purpose, I suggest you to change "Taux de graisse corporelle" by "Pourcentage de graisse corporelle"
- you can translate "Bone mass" by "Masse osseuse"
- You can translate "Basal metabolic rate" by "Taux métabolique basal (BMR)" (keep BMR into brackets as it's also commonly used in France).

My scale is also giving the Visceral fat (graisse viscérale). Don't know if it's relevant for open scale (kind of indicator between 0 to 60). Look at https://tanita.eu/tanita-academy/understanding-your-measurements

Hope it helps.

Fabrice

oliexdev, 2017-11-19:
thanks I added yours French translations but still some French translations are missing.
Yeah visceral fat is another measurement see #4

fabads, 2017-11-19:
OK I will check and translate it for you

oliexdev, 2017-12-05:
I will close this issue here. If somebody has more French translations please create a PR or comment it here. Thanks in advance.

closed 2017-12-05 by oliexdev