oliexdev/openScale #103: Could you help me with connecting to Renpho scale with openscale.

Lakshmi-Narasimhan, 2017-11-22:
Am a newbie and am unable to understand reverse engineering renpho scale as this scale is not even pairing with my bluetooth, the scale is connected and  values are shown only on the renpho app.

oliexdev, 2017-11-25:
Did you try to reverse engineer your scale as described in the [wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Blueeoth-4.x-scale)?

maszek, 2017-11-29:
That's a complex process. It's not clear where certain values should be filled in. 
The Bitsnoop log file is not readable:

[image]

The Open Scale app is searching, but only for a particular type of scale (MI-scale), it seems:

[image]

Finally, the method described still requires an app from Google Play (BLE scanner). I want to stay away from that.

P.S. is "Blueeoth" a typing error?

oliexdev, 2017-12-01:
@maszek you have to open the Bootsnoop log file with [wireshark](https://www.wireshark.org/) 
openScale is searching for default the Mi scale but if you device is not supported then openScale can't connect to your scale.

Maybe they exist another BLE scanner that are open source.

Yeah Blueeoth was of course a typing error thanks for your hint.

maszek, 2017-12-06:
Okay, I was hoping it would be easier when I read the 'crate of beer' instruction, which is about the level I am able to comprehend.
Good luck developing your app, it looks promising, I'll check in again later.

oliexdev, 2017-12-06:
ok no problem I will close this issue. Please create a new issue or comment here if you have further questions related to this topic.

closed 2017-12-06 by oliexdev