oliexdev/openScale #108: [Help] Default csv format

giuaig, 2017-11-29:
Hello,
i'd like to import old data from a Sanitas SBF70 scale and their Healthcoach app.

I have found this columns in the old app database:
`ScaleMeasurementID,UserId,MeasurementDate,MeasurementTime,WeightKg,WeightPound,BodyFatPct,BodyFatPctUpper,BodyFatPctLower,WaterPct,MusclePct,MusclePctUpper,MusclePctLower,BoneMassKg,BoneMassPound,BMI,BMRKCal,AMRKCal,ActivityGrade,Comment,IncludeInGraph,IsAddedManually,IsUpdatedManually,DeviceId,CreatedDate,UpdatedDate,DeletedDate,Revision,IsDeleted,GlobalTime,IsNewRecord,Impedance,Source,UpdatedSource,DeviceClientRelationshipId`

Where can i found which columns openScale uses and its order?

erijo, 2017-11-29:
See the importData method in https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/OpenScale.java

oliexdev, 2017-12-01:
as @erijo already said it is in the above java file at `importData(..)` defined.

your csv format hast to be look like:
`DateTime;Weight;Fat;Water;Muscle;LBW;Bone;Waist;Hip;Comment;`

The date time format should look like `dd.MM.yyyy HH:mm`

But the simplest method is to export some sample data an look at the generated csv file.

closed 2017-12-01 by oliexdev

giuaig, 2018-03-27:
Sorry if i re-open this thread, i have created a script to migrate from HealthCoach app to openScale, maybe it could be useful to other people, here is the link:
https://github.com/giuaig/sanitas2openscale/blob/master/sanitas2openscale.py

Thank you devs for your hard job and the time that you spend in openScale project ;)