oliexdev/openScale #111: Need Help: reverse engineering Beurer BF 700 bt protocol

linuxlurak, 2017-12-02:
Hello! I have a Beurer BF 700 and the bulky "HealthManager App" from Beurer for Android. I decoded the Android App in a few steps and got the source code. But I don't know what to look for. 

I found the fcc docs from beurer here: https://fccid.io/VGR-BF700.

Publishing the app's source code would be an infringement; it is closed source. Please contact me privately for more info about the source code.

Have a peek in the files here:
[image]

and here:
[image]

The BF700 device is generically handled as a BF800.

oliexdev, 2017-12-03:
The FCC documents are not relevant. Decompiled source code is hard to understand.  Please read the [wiki page](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) for how to reverse engineer the Bluetooth protocol. Btw I can't contact you because you don't have any email address published on your github page.

Dododappere, 2017-12-15:
The Beurer BF700/800 scales use basically the same protocol as the Sanitas SBF70. Sanitas is actually a sub-brand of Beurer...

Where Sanitas commands start with 0xe7, Beurer commands start with 0xf7. The byte that follows after that is usually the same for both the Sanitas and Beurer scales, apart from some exceptions.

linuxlurak, 2017-12-15:
I assumed this too! How did you get proof of it? How do you know the start command? It compared detail photos of the devices, but had no access to the internal electronic or even a firmware dump...

I'll check it with the start command, thanks for letting us know!

linuxlurak, 2017-12-15:
Some days before I forked openscale and created a Beurer BF700 clone based on the sanitas here: https://github.com/linuxlurak/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothBeurerBF700.java

I'll compile the app and post the results here if it works with your starting byte.

linuxlurak, 2017-12-15:
connecting to the beurer bf700 works according to the app:
[image]

No data gets synchronized at the moment. I assume the app requests the data on every connection with the scale?

Perhaps wrong UUID etc. I have to check on that.

Dododappere, 2017-12-15:
From looking at your code there are a couple of more changes needed:
- Change 0xe6 to 0xf6
- Change 0xe9 to 0xf9

For the rest I suggest you step through it with a debugger to see what works and what doesn't...

linuxlurak, 2017-12-15:
@Dododappere so many thanks to you! It works!

Dododappere, 2017-12-15:
You're welcome!

The Beurer BF800 will also work with this code...how about enabling that one as well?

linuxlurak, 2017-12-15:
Done!

oliexdev, 2017-12-16:
thanks @linuxlurak and @Dododappere 

Could you tell us what is working and what not, so I can write the correct entry in our supported scale table (https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale)

Additionally could you post an image of your scale for our gallery (same angle like the other ones)

Thanks in advance

linuxlurak, 2017-12-16:
Pictures:
[image]
[image]

AMR is not displayed in the app only BMR. At the moment I know nothing about user creation, multi users, etc.

closed 2017-12-17 by oliexdev

oliexdev, 2017-12-17:
Could you tell me how your Beurer device name is written in the android application? Can you test the version 1.6.1? One user reported that his Beurer scale is not supported.

linuxlurak, 2017-12-17:
[image]

linuxlurak, 2017-12-17:
Indeed c 1.6.1 does not recognize my BF700. See my commment on your commit https://github.com/oliexdev/openScale/commit/f465e4e8d9eecc5d2c5cfdaecd89f237db1ecd49

oliexdev, 2017-12-17:
I fixed it with commit https://github.com/oliexdev/openScale/commit/f465e4e8d9eecc5d2c5cfdaecd89f237db1ecd49 
I already updated all released files on GitHub and GooglePlay. Please test the new file.
Under version on the about page in openScale should now be displayed "v1.6.1 (21)"

linuxlurak, 2017-12-17:
Build 21 works with BF700

Dododappere, 2017-12-20:
Hi guys, today I got my hands on a Runtastic Libra which looked suspiciously similar to the BF800. So I did some tests and it behaves EXACTLY the same as the BF800!

It advertises itself as "RT-Libra-B". Can you add it as well?

erijo, 2017-12-20:
I can't find 1.6.1 on Google play (nor on f-droid).

oliexdev, 2017-12-20:
@linuxlurak  thanks unfortunately an user still reported a problem as follow:

> Updated app again, now Beurer BF700 is seen and accepted but has problem to connect and doesent load data

But as he report it via a two star review I can't really help because his problem description is not really helpful and the review system on GooglePlay is not really designed for a discussion.

@Dododappere thanks so did you already test the Runtastic libra scale with openScale?

@erijo On GooglePlay I only rolled out the new version to 50% of the users for testing purpose (I will change that in a few day). On f-droid I have no control when they update their server.

linuxlurak, 2017-12-21:
@oliexdev hmm the device connects and it shows the weight and it's changing values when I step on the scale.

I created a new user. this user was added to the scale and the data gets synchronized. 

My first user was not able to synchronize anymore. That's true; I had the same problem as the cited user. Is the sync process in anyway scale dependent? So the change from BF700 to BF700/800 could have an impact?

If you want, I'll test it some more. 

Can you tell my the following, so I can test more specifically:
- when exactly gets the data synchronized? On every successfull BLE connection with the scale? So once per connection?
- if I delete a user on the app, is the user deleted on the scale too?
- As I understand it, the initial weigh (create new user, step on scale to have a comparable weigh) is only done once after creating a user, correct?

linuxlurak, 2017-12-22:
My scale tells me that it's full. OpenScale seems not to clear the already synchronized data. At the moment I can't take further test. I have to install the beuerer app first and delete all the data.

Dododappere, 2017-12-22:
@linuxlurak there is a method deleteScaleData() in the code for the BF700 which should delete all the stored measurements for a user. Is that not called? Perhaps add some extra logging to see if that is called or not. However, if you have other users that are also still using the scale, the memory will obviously be full at some point....openScale does not delete measurements from other users.

There is one issue with the code regarding the use of the currentScaleUserId. The code only seems to ask for measurements for currentScaleUserId. But if you previously used the scale with the Beurer app, it will only work if the currentScaleUserId matches the one already on the scale. The current implementation tries to match the first 3 characters of the logged-in openScale user with what is stored in the scale. That can work in some cases but it is not fool proof....are you sure you have the right user selected?

@oliexdev I'll see if I can install Android Studio during Christmas and see if I can compile openScale. I have an old Nexus 5X somewhere in the drawer and perhaps I can do some testing with Runtastic Libra. I don't have any Android experience but I'll see how far I get...

Dododappere, 2017-12-22:
@oliexdev I am trying your app (from Google Play store) with my Beurer BF800 and it doesn't work. It sees the scale but it says 'Beurer BF800-B [device not supported]'. I think it is because of the '-B' in the advertisement name....

linuxlurak, 2017-12-23:
@Dododappere can you instruct me, where i can find the log? a log of android OS? In the source are some writings to a log file but I don't know where this log file is located...

oliexdev, 2017-12-23:
@Dododappere 
the method is only called if the current item reached the maximum item, see 
```
            if (current_item == max_items) {
                // finish and delete
                deleteScaleData();
            }
```

Thanks, good to know if the Beurer use also another Bluetooth device name as -B in the name. I changed the device name checking with commit https://github.com/oliexdev/openScale/commit/7283bd17335bd8e7a77aca02a4cbd18884140bbd

Should work now with the latest source code. Thank your for considering installing Android Studio if you have question how it works in Android don't hesitate to ask.

@linuxlurak 
the logs are only displayed as output messages no log file are written.
They easiest way to see it, is to use [ Android Studio logcat](https://developer.android.com/studio/debug/am-logcat.html). 

Alternative you can use the command line [tool adb](https://developer.android.com/studio/command-line/logcat.html).
Or if you using Linux

linuxlurak, 2017-12-23:
So i filled up my BF700 with measurements till it displayed "FULL". Now no data gets synced at all. My scale stays full. Perhaps this case is not handled?

I now compiled the newest github version of openscale. I'll try to add so more log output and report back.

linuxlurak, 2017-12-23:
I added 
`Log.v(TAG, "logtest");`
to some points of the code but I can't find anything in logcat... I am lost.

erijo, 2017-12-23:
Perhaps that log level is filtered out. Try Log.d or Log.e instead.

linuxlurak, 2017-12-23:
log level error seems to work. How do I switch on loglevel debug? all the log output of openscale is on level debug. That's why I have zero output...

erijo, 2017-12-23:
See https://developer.android.com/studio/command-line/logcat.html#filteringOutput

oliexdev, 2017-12-23:
@linuxlurak see http://adbshell.com/commands/adb-logcat

linuxlurak, 2017-12-23:
@oliexdev & @erijo thank you for trying to help me! ;D

[image]

As I understand it (both of your links) this prints the log according to the log level. But I have nothing that gets printed to the log file at all... So it's not a problem of sorting / filtering the log file but of having anything from the app in the logs at all. Do you see my point or I am totally wrong?

linuxlurak, 2017-12-23:
At the shell (adb logcat) the app prints nothing. Isn't it the same log in android studio anyways?

erijo, 2017-12-23:
Are you running the debug build? Otherwise debug logs may be removed.

I use the "Run" tab in Android studio to see debug logs. Perhaps you can try that?

linuxlurak, 2017-12-23:
I normally press the debug button little gear icon. Even with menu Run> Debug 'app' Log.d won't show up. But Log.e works. I don't know what setting to set where. I'll replace all Log.d with log.e for the moment. Thanks for your effrots @erijo !

linuxlurak, 2017-12-23:
Ok. Now there is some output:

> 12-23 19:16:24.519 4904-4917/com.health.openscale E/BEURER BF700/800: Write new Date/Time:1514052984 00 00 00 00 5A 3E 9D 78 
> 12-23 19:16:24.689 4904-4916/com.health.openscale E/BEURER BF700/800: ACK Scale is ready
> 12-23 19:16:25.239 4904-4977/com.health.openscale E/BEURER BF700/800: ACK Got general user information
> 12-23 19:16:25.239 4904-4977/com.health.openscale E/BEURER BF700/800: Count:2 maxUsers:8
> 12-23 19:16:25.240 4904-4977/com.health.openscale E/BEURER BF700/800: Ack Get UUIDSs List of Users
> 12-23 19:16:25.244 4904-4977/com.health.openscale E/BEURER BF700/800: Send ack gotUser
> 12-23 19:16:25.338 4904-4917/com.health.openscale E/BEURER BF700/800: Ack Get UUIDSs List of Users
> 12-23 19:16:25.341 4904-4917/com.health.openscale E/BEURER BF700/800: Send ack gotUser
> 12-23 19:16:25.343 4904-4916/com.health.openscale E/BEURER BF700/800: Request getUserInfo 101
> 12-23 19:16:25.344 4904-4916/com.health.openscale E/BEURER BF700/800: scaleuserid:101 registered users: 2 extracted users: 2
> 12-23 19:16:25.438 4904-4977/com.health.openscale E/BEURER BF700/800: Ack Get User Info Initials
> 12-23 19:16:25.438 4904-4977/com.health.openscale E/BEURER BF700/800: Name [DETAILS CUT OUT]
> 12-23 19:16:25.441 4904-4977/com.health.openscale E/BEURER BF700/800: Request Saved User Measurements
> 12-23 19:16:25.538 4904-4917/com.health.openscale E/BEURER BF700/800: Ack Get scale status
> 12-23 19:16:25.539 4904-4917/com.health.openscale E/BEURER BF700/800: BatteryLevel:94 weightThreshold: 2.0 BodyFatThresh: 2.0 Unit: 1 userExists: true UserReference Weight Exists:true UserMeasurementExists false scaleVersion6
> 12-23 19:16:42.889 4904-4916/com.health.openscale E/BEURER BF700/800: DataChanged - not handled: E0 01

Dododappere, 2017-12-23:
@oliexdev Confirmed, the Runtastic Libra indeed works! Same as BF800.

oliexdev, 2017-12-27:
@Dododappere thanks!