oliexdev/openScale #115: Cannot delete Beurer BF700 measurements

igstewart, 2017-12-22:
Readings from Beurer BF700 scales with 1.6.1 is recording the record id value as NULL.

This is resulting in readings from the scale having no trash can to delete them.

Exporting a backup, manually editing the record id and importing resolves the issue.

oliexdev, 2017-12-23:
what do you mean with the record id? Is the record id from the scale? What do you mean it has no trash can?
Sorry currently I can't follow you but I also don't own this scale. Maybe the problem is along with the issue #111 which is currently in discussion?

igstewart, 2017-12-23:
When you backup data from the app it saves a database and there is an id field which increments with each new reading, this has a NULL value for readings from the scale.
[image]

When manually inputting weight a trashcan is shown to delete individual records.
[image]

When read from the scale it is not there.
[image]

linuxlurak, 2017-12-23:
I reinstalled openscale (from playstore this time) and reset my scale. Ill report back when the scale gets full again and won't delete the old results. Now jumping on and off the scale ;)

igstewart, 2017-12-23:
There doesn't seem to be any issue in readings being deleted from the scale itself for me.

It's purely removing them from the app itself I'm having issues with.

linuxlurak, 2017-12-23:
So i filled up my BF700 with measurements till it displayed "FULL". Now no data gets synced at all. My scale stays full. Perhaps this case is not handled?

igstewart, 2017-12-23:
This is a different issue from mine...

linuxlurak, 2017-12-23:
Sorry. You specifically meant the delete button / trash can, I see now, right? I'll open a new issue.

oliexdev, 2017-12-23:
@igstewart Thanks Now I understand what you mean. Unfortunately I can't reproduce this behaviour. The id is defined as a primary key in the sql database, so its automatically set by the database. Also the manual adding via the add dialog and the automatic adding of the Bluetooth data is done with the same method `addScaleData(..)` in openScale.java file. 

Could it be that you are using an OnePlus smartphone?

igstewart, 2017-12-23:
I'm using a Xiaomi Mi 5 running Lineage 14.1

oliexdev, 2017-12-23:
hmm sorry I don't know I can't find the problem. Maybe its help to reinstall openScale? (Make before a backup of your data)

igstewart, 2017-12-23:
There seems to be some sort of issue with exporting/importing backups.

I wiped the app data and set everything up fresh and readings from the scale are able to be deleted.

If I then export a backup of these readings and import it back again the  records lose the ability to be deleted again.

So a fresh, known working backup stops working when it is imported, never mind the one I made to import my beurer app data.

oliexdev, 2017-12-23:
did you change something (e.g. add new rows) in the exported database?
Normally it can't break the database by exporting and importing the database again because it is an exact copy of the database.

If you want to import your old data from the beurer app it is better to use the import function on the table page. Create a csv file in the same format for openScale. See how it should look like by exporting a sample into a csv file.

igstewart, 2017-12-23:
OK, I got it working by just doing a csv import, thanks!

But there is definitely something wonky with the backup import/export function in the settings menu.

As I say, exporting and immediately re-importing data purely created in app leads to breakage.

oliexdev, 2017-12-23:
that is strange but currently I can't reproduce this behaviour.

closed 2017-12-23 by oliexdev

Dododappere, 2017-12-23:
@oliexdev I managed to compile openScale and running it. I can confirm the Beurer BF800 works with your last fix!

One question though, I had to try a couple of time before it would actually connect to the device. I saw in the log some lines with 'No device found'.....I am running Android 8.1 on a Nexus 5X. Does this sound familiar? It seems reconnecting to a known device is a bit unreliable...

oliexdev, 2017-12-27:
@Dododappere 
It could be happened if the last Bluetooth connection was not successful terminated or if you reconnected to quickly. Currently I don't have a solution to this problem. :(