oliexdev/openScale #125: Not possible to import backup created with 1.6.1

erijo, 2018-01-02:
A backup created with 1.6.1 can not be imported when running latest master due to the room change.

erijo, 2018-01-03:
It doesn't seem like it possible to import a backup created with the latest master either.

oliexdev, 2018-01-03:
I know but the developer version is not the final version. Changing the database framework is not a small step. There is still some work to do ;)
Currently I am experimenting with the openCSV library for exporting / importing to/from a CSV file but I am getting stuck with the library.
I want to improve the quality of the source code that it will be easier to maintain and to have less work in the future to implement a new measurement.

erijo, 2018-01-03:
Ok, I see. Will hold my horses reporting bugs then :)

Given #122 and #126, do you think it would be a good idea to create a 1.6.2 release based on 788abef? If you think so and want some help I could help prepare a branch for it. Just let me know.

oliexdev, 2018-01-04:
That was not my intention ;) You are welcome to report more bugs and issues.

For the next release some issues should be fixed first:
I want to get some positive feedbacks weather the SBF 75 scale is working or not, see #118.
Then the implemented Room data migration strategy needs more testing so that no data will be lost on a upgrade.
And last but not least I want to implement a better CSV import / export function which makes it more easier to use. But currently I am struggling with every CSV library because of a gradle build error:

> Didn't find class "java.beans.Introspector" on path: DexPathList

It seems various CSV libraries have always a conflict with another library. :thinking: 

An official release only to fix the date calculation bug for cosmetic reasons is inappropriate. If this were a major bug so you wont able to use openScale any more then I would agree with you. You need to consider that on each upgrade every openScale user gets an notification and they need at least some bandwidth on mobile devices. And if you keep doing unnecessary updates, I can imagine that the users will get annoyed.

oliexdev, 2018-01-05:
I implemented a new CSV import / export method with the SimpleCSV library and updated the export/import methods. 
But the underlying data structure had been changed a lot so there is no backwards compatible with any old CSV or backup data of previous openScale versions.

closed 2018-01-05 by oliexdev