oliexdev/openScale #129: Measurements unit

woj-tek, 2018-01-05:
Hi,
thanks for the great project! I've been using openScale with my non-IoT scale for more than a year and it was great.

Recently I've got xiaomi mi scale 2 and.. for some reason it gives the muscle measurements in kilograms instead of % (and also is about 20 % points off of my previous scale...), so if I wan't to put that into openScale I would have to calculate that manually.

**tl;dr**
please add toggle to input measurement unit in [kg] or [%]

oliexdev, 2018-01-05:
I understand that its similar to PR #59 and issue #4. 
As in both links described currently It is not so easy to add some new measurement in a clean way.
But step by step its getting easier. I changed the internal database library and so on but there is still some work to do.

closed 2018-01-05 by oliexdev

woj-tek, 2018-01-05:
more to #59 - I've missed that one (and I've saw #4 but IMHO this is slightly different). I've commented there.